JdeRobot/RoboticsAcademy #1990: [React frontend] Play button shouldn't be clickable while the program is running

Rogarsu2014, 2023-03-29:
While the code is running the "Play" button should be deactivated, whereas the "Pause" and "Stop" buttons should stay activated.

This is because clicking "Play" makes it so that methods such as "load_code" and "run" are activated without stopping the previous code first, which can cause problems.

Bhardwaj-Himanshu, 2023-04-05:
Hi @Rogarsu2014,
Looking onto this issue, it seems very valid to disable a button while other managing processes required to execute it are running.

There could be more than 2 methods for this, but these 2 came to my mind-
-Either we can disable the "Play" button, when clicked, by setting a 'button_click' property to true and false
- or we can use Switch cases with tagging things as background worker.

Also, do update me if you have arrived on something!

ango1994, 2023-04-18:
Solved on #1966

closed 2023-04-18 by ango1994