JdeRobot/RoboticsAcademy #1997: [3d_reconstruction] Error loading kobuki plugin

pariaspe, 2023-04-02:
```
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_kobuki.so: libgazebo_ros_kobuki.so: cannot open shared object file: No such file or directory
```

saurabhmj11, 2023-04-04:
The error message you are seeing suggests that Gazebo is unable to load the plugin libgazebo_ros_kobuki.so because it cannot find the shared object file. Here are some steps you can try to fix this issue:

Verify that the plugin is installed: Check if the plugin libgazebo_ros_kobuki.so is installed in your system. If it is not installed, you will need to install it.

Check if the plugin is in the correct directory: Verify that the plugin libgazebo_ros_kobuki.so is located in the correct directory where Gazebo is looking for plugins. By default, Gazebo looks for plugins in the /usr/lib or /usr/local/lib directories. If the plugin is not in one of these directories, you will need to move it to the correct directory.

Check if the plugin path is set correctly: Check if the environment variable GAZEBO_PLUGIN_PATH is set correctly to include the directory where the plugin libgazebo_ros_kobuki.so is located. You can check the current value of the GAZEBO_PLUGIN_PATH variable by running the following command: echo $GAZEBO_PLUGIN_PATH. If the directory where the plugin is located is not included in the output, you will need to add it to the GAZEBO_PLUGIN_PATH variable.

Rebuild the plugin: If the above steps do not work, try rebuilding the plugin libgazebo_ros_kobuki.so from source. This can help in resolving any compatibility issues with your system.

Check for errors: Check the console or error logs for any errors or warnings related to the plugin. Fix any issues that might be preventing the plugin from loading properly.

If none of the above steps work, try providing more details about your system and the Gazebo installation, including any error messages or warnings you receive. This will help in identifying the specific issue and providing a more targeted solution.

Bhardwaj-Himanshu, 2023-04-05:
Hi @pariaspe,
-Can you also mention the steps to reproduce this issue!

The title says Error loading kobuki plugin so where can I find it?

pariaspe, 2023-04-05:
Hi @Bhardwaj-Himanshu ,

To reproduce this issue try to run `3d_reconstruction` exercise. That message should appear in your terminal.

It seems that the package [kobuki_gazebo_plugins](http://wiki.ros.org/kobuki_gazebo_plugins?distro=kinetic) is missing in the RADI. Try to add it to the dockerfile, build a new image and check if it is still missing.

Bhardwaj-Himanshu, 2023-04-05:
Hi @pariaspe,
Tried launching the `3d_reconstruction` exercise. Not that message, but a similar or same error from #1904 was raised.
I am attaching a screenshot below to confirm it, could be an error on my side.
Tried launching 2-3 instances one after other, after successfully closing each one of them, so that nothing was conflicting with each other.

Do let me know if this got raised on your end too!
I'll update you soon on the kobuki_gazebo_plugins soon!

[image]

pawanw17, 2023-04-10:
@pariaspe unfortunately the package https://github.com/yujinrobot/kobuki_desktop doesn't support Noetic, and hence it is not there in PPA.

GauravKumar9920, 2023-04-10:
Hey @pawanw17 , the kobuki plugin supports ROS Noetic [1], check it's on a different branch. you can check the documentation [2]

[1] - https://github.com/yujinrobot/kobuki_desktop/tree/kinetic
[2] - http://wiki.ros.org/kobuki_gazebo

pawanw17, 2023-04-10:
@GauravKumar9920 it only supports till melodic, there isn't a branch for noetic...

pariaspe, 2023-04-11:
Hi @pawanw17 , can you paste those issues here?

pawanw17, 2023-04-13:
Hi @pariaspe Here's what I did:

I added kobuki_gazebo and kobuki_gazebo_plugins from https://github.com/yujinrobot/kobuki_desktop in /catkin_ws/src and did a catkin build, since these packages are cpp based there wasn't an issue in building them.

Here's what I observed:
There's an issue where the gazebo screen is black and doesn't load up the models.

[image]

pariaspe, 2023-04-13:
So far so good, any idea which models are causing the delay problem? Does the terminal show any message about the models?
As far as I know the models should be at CustomRobot repo but they can be missing

pawanw17, 2023-04-17:
@pariaspe the terminal doesn't show any message about the missing models, I am looking into which models are being downloaded/causing delay...

pawanw17, 2023-04-17:
@pariaspe The above two MRs should close this issue.

pariaspe, 2023-05-16:
Fixed