JdeRobot/RoboticsAcademy #1999: [Drone Hangar Exercise] Unable to launch Drone Hangar exercise correctly

pawanw17, 2023-04-06:
To reproduce just launch the (drone hangar exercise)[https://jderobot.github.io/RoboticsAcademy/exercises/Drones/drone_hangar] in the RADI.

Errors in logs:
[image]
 
[image]

The issue is likely related to Gazebo not launching correctly and crashing.

pariaspe, 2023-04-17:
Not finding cylinder model?
[image]

pawanw17, 2023-04-18:
Hi @pariaspe the issue here is of missing models, 
Here's what happens

1. The first Gazebo instance runs 
2. Waits to download the models 
3. Timeout occurs
4. That's when a second instance is started by launch.py
5. The two instances eventually crash each other.

pawanw17, 2023-04-18:
The exercise tries to download these models but is unable to find them
[image]

pariaspe, 2023-04-18:
Got it @pawanw17 , the problem is this unmerged [PR](https://github.com/JdeRobot/drones/pull/183). The thing is that some parts of the PR are wanted and others unwanted. We still have to modify some parts of it.

pariaspe, 2023-05-26:
Solved in JdeRobot/drones#188

closed 2023-05-26 by pariaspe