JdeRobot/RoboticsAcademy #2006: Catkin Build failed packages in RADI

pawanw17, 2023-04-17:
The catkin build command inside the RADI results in the failure of these 4 packages:

1. dataspeed_can_usb
2. dataspeed_can_tools
3. dbw_mkz_joystick_demo
4. dbw_mkz_twist_controller

Along with 20 packages which are abandoned during the build

[image]
[image]

If any exercise uses these, then it may have issues else if these are not needed anymore then can be removed.

pariaspe, 2023-04-17:
Hi @pawanw17 , I'm able to build the RADI without facing that error. Could you provide some extra indications to reproduce the error?

Amglega, 2023-04-18:
Not really, they are in the same directory as the Catvehicle package but are not used in the autoparking_v2 exercise because the car does not need to be teleoperated.

closed 2023-05-15 by pariaspe