JdeRobot/RoboticsAcademy #2007: [Human Detection] Console not opening

pariaspe, 2023-04-17:
```
Exception happened during processing of request from ('172.17.0.1', 55732)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.8/dist-packages/websocket_server/websocket_server.py", line 271, in __init__
    StreamRequestHandler.__init__(self, socket, addr, server)
  File "/usr/lib/python3.8/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.8/dist-packages/websocket_server/websocket_server.py", line 284, in handle
    self.read_next_message()
  File "/usr/local/lib/python3.8/dist-packages/websocket_server/websocket_server.py", line 341, in read_next_message
    opcode_handler(self, message_bytes.decode('utf8'))
  File "/usr/local/lib/python3.8/dist-packages/websocket_server/websocket_server.py", line 163, in _message_received_
    self.message_received(self.handler_to_client(handler), self, msg)
  File "/RoboticsAcademy/exercises/static/exercises/human_detection/exercise.py", line 549, in handle
    start_console()
  File "/RoboticsAcademy/exercises/static/exercises/human_detection/console.py", line 11, in start_console
    sys.stderr = open('/dev/pts/' + console_fd, 'w')
OSError: [Errno 5] Input/output error: '/dev/pts/1'
```

ammarfitwalla, 2023-04-18:
Hi , I am new to open source , can I have this issue resolved?

pariaspe, 2023-04-19:
Hi, go ahead and propose a pull request with your solution. Also, take a look at the [contributing guide](https://github.com/JdeRobot/RoboticsAcademy/blob/master/CONTRIBUTING.md) please.

Bhardwaj-Himanshu, 2023-05-01:
Let me know what you are upto @ammarfitwalla , and if I could be of any help!

pawanw17, 2023-05-10:
@pariaspe  The issue occurs because the console is open at a very early stage of the code, moving to a later stage (similar to other exercises) fixes it.
[image]

PR #2053 (2023-05-11): moved start_console to a later stage
Fixes #2007 by starting the console at a more appropriate place