JdeRobot/RoboticsAcademy #2027: Broken forum hyperlink on homepage

pawanw17, 2023-04-29:
[image]
The button redirects to 
https://forum.jderobot.org/c/english/roboticsacademy/12
which is deprecated, instead it should redirect to
https://forum.unibotics.org/

Bhardwaj-Himanshu, 2023-04-30:
Hi @pawanw17,
This issue was an easy fix, just located the forum links and changed it to the provided forum link.

But I am stuck at `**Github/Git adding multiple files to the pull request, when I only want to add one.**`

To resolve this issue `I created a new branch, added the changes to it, commited it, and pushed it to my fork repo as another branch`, but it is still showing me other commits in other branches as addition.
No idea on what to do?
Just a small guess of doing git reset --hard SHA<id>, but if anything else that could solve this?

It looks like this-
[image]
[image]

jmplaza, 2023-04-30:
Fixed in #2028

pawanw17, 2023-04-30:
Hi @jmplaza the issue still exists... I suspect the wrong file was changed...

jmplaza, 2023-04-30:
It has been changed at the repo, but no RADI has been released yet with it. So it will be solved in the next RADI release.

pawanw17, 2023-04-30:
I implemented his changes locally to test... I think the correct place should have been :    
https://github.com/JdeRobot/RoboticsAcademy/blob/3616a42b394989c9c2368fe4492bdc97a8d55834/react_frontend/src/components/DrawerAppBar.js#L28

jmplaza, 2023-04-30:
perfect, just go ahead with the PR and you will solve this minor issue.

pawanw17, 2023-04-30:
Hi @Bhardwaj-Himanshu Do you want to take this up?

Bhardwaj-Himanshu, 2023-04-30:
I would love to @pawanw17 , but you can contribute to it as well and let me know what are you upto!

pawanw17, 2023-04-30:
Hi @Bhardwaj-Himanshu Since you're already working on this, you can continue and finish it. I'd just suggest you test your changes before making the PR. All the best!

Bhardwaj-Himanshu, 2023-04-30:
Yup my bad, will keep in mind next time to create a RADI image as well, coming onto the issue of RADI image, I am still stuck on--

Could you take a look at this, whenever you are free? Much appreciated mate.
https://github.com/JdeRobot/RoboticsAcademy/issues/1904#issuecomment-1528927544

closed 2023-05-12 by ango1994