JdeRobot/RoboticsAcademy #2041: [ Exercise ] New Image Processing exercise.

felicidadaqm, 2023-05-03:
Issue to upload the new exercise of Image Processing introduction level.

onkar-1432, 2023-12-22:
HI @felicidadaqm can you describe these issue with more specifications.

jmplaza, 2025-03-20:
The [Basic Computer Vision exercise](https://jderobot.github.io/RoboticsAcademy/exercises/ComputerVision/basic_computer_vision), with real cameras, has been completely refactored using WebRTC technology to get the camera frames from the browser. This way it works not only for Linux but also for Windows and MacOS machines. The camera support using ROS drivers inside the RADI or RoboticsBackend docker container works for Linux, but not for Windows machines. It is due to security reasons preventing the mapping of USB ports from the docker container in Windows machines.

Kudos to @codezerro @jessiffmm @vmartinezf @javizqh and @lu164 ! :-)

closed 2025-03-20 by jmplaza