JdeRobot/RoboticsAcademy #2178: Dependencies_noetic docker build crashing

pawanw17, 2023-07-16:
Failing at this line
https://github.com/JdeRobot/RoboticsAcademy/blob/30178b55910840d0d002e0117657f6242f0cc6ba/scripts/mini_RADI/Dockerfile.dependencies_noetic#L55

Turbovnc has moved

Bhardwaj-Himanshu, 2023-07-22:
Hi @pawanw17 , replacing it with a particular version would cause us to replace it every single time a new version is released!

But the general link as mentioned by you above is failing, could we replace it with something else in a longer run?And sticking to the 3.0 for now.

pawanw17, 2023-07-22:
@Bhardwaj-Himanshu the version we use has been 3.0 for a while, and hasn't been upgraded to the latest version (I don't think we require an upgrade), I suggest to have .deb file inside the repository, the way it is done for Humble RADI, you can make a PR for this...

Bhardwaj-Himanshu, 2023-07-22:
I have created a pull request, should we update the line after as well?
https://github.com/JdeRobot/RoboticsAcademy/blob/30178b55910840d0d002e0117657f6242f0cc6ba/scripts/mini_RADI/Dockerfile.dependencies_noetic#L56-L57
and if not, could you tell me on why?

pawanw17, 2023-07-28:
Hi @Bhardwaj-Himanshu I meant to have the deb file locally saved in the gpu folder and change the install instructions in the Dockerfile making them similar to how Humble dockerfile installs turbovnc,

Tanishq30052002, 2024-03-05:
Hi @pawanw17 ,
It seems the issue is still there, since currently also using that same link in the master branch. So should it need a correction?

jmplaza, 2024-07-01:
Noetic branch is now deprecated, and ROS2 Humble is the only supported release now

closed 2024-07-01 by jmplaza