JdeRobot/RoboticsAcademy #2239: Optimization of miniRADI docker image size

jmplaza, 2023-09-13:
Currently it is very large, 4.4GB compressed. It includes all the dependencies of robotics applications, but maybe its size can be reduced somehow.

pariaspe, 2023-10-05:
Some related issues with recent related work:
- https://github.com/JdeRobot/RoboticsInfrastructure/issues/276 
- https://github.com/JdeRobot/RoboticsInfrastructure/issues/315
- https://github.com/JdeRobot/RoboticsInfrastructure/issues/277
- https://github.com/JdeRobot/RoboticsInfrastructure/issues/280
- https://github.com/JdeRobot/RoboticsAcademy/issues/2227

Surajpatra700, 2024-03-08:
Sir @jmplaza i tried to build docker image of "Dockerfile.dependencies_humble" file using command "sudo docker build -t mini_radi ./Dockerfile.dependencies_humble ." it give me this error:
[image]
I searched it over Internet but unable to get a solution that worked for me.

pariaspe, 2024-03-11:
Hey @Surajpatra700,
We have a script in that folder to build RA docker images, try it instead of building manually via cli. Anyway, you are using wrongly the command, I would recommend you to check the docker official documentation.
Btw, you shouldn't use `sudo` before docker commands. Take a look to that too.

jmplaza, 2025-05-16:
https://hub.docker.com/r/jderobot/robotics-academy/tags

Sarvesh-Mishra1981, 2026-02-15:
hello @jmplaza, Can i contribute to this.

Dev-sh21, 2026-03-06:
Thanks for the clarification @javizqh sir,
 that makes sense, I’ll keep the DL stack requirement as it is.

While diving the rest of the image, i noticed that CustomRobots alone takes about 2.6 gb. I initially thought this might be .git history or build artifacts, but it seems to be mostly from the repository itself, like the  marker_visual_loc alone is close to 1 gb.

Are there specific large assets in these packages that are required for the simulations, or could some of them be legacy or optional assets?

I just want to better understand what contributes to this 2.6gb so i can identify safe opportunities for reducing the image size