JdeRobot/RoboticsAcademy #2253: [Humble] Rescue people when launched without --debug flag gives No module error

pawanw17, 2023-09-25:
```shell
2023-09-25 05:30:34,478 [MainThread  ] [INFO ] (root)  Launch transition started, configuration: {'application': {'type': 'python', 'entry_point': '$EXERCISE_FOLDER/entry_point/ros2_humble/exercise.py', 'params': {'circuit': ''}}, 'launch': {'0': {'type': 'module', 'module': 'drones_ros2', 'resource_folders': ['$EXERCISE_FOLDER/launch/ros2_humble'], 'model_folders': ['/catkin_ws/src/drones/drone_assets/models/boat_beacon'], 'plugin_folders': [], 'world_file': '$EXERCISE_FOLDER/launch/ros2_humble/world.json'}}, 'exercise_id': 'rescue_people_newmanager', 'visualization': 'gazebo_rae'}
Traceback (most recent call last):
  File "/RoboticsAcademy/exercises/static/exercises/rescue_people_newmanager/entry_point/ros2_humble/../../python_template/ros2_humble/exercise.py", line 21, in <module>
    from hal import HAL
  File "/RoboticsAcademy/exercises/static/exercises/rescue_people_newmanager/python_template/ros2_humble/hal.py", line 5, in <module>
    from jderobot_drones.drone_wrapper import DroneWrapper
  File "/home/ws/build/jderobot_drones/jderobot_drones/drone_wrapper.py", line 9, in <module>
    from as2_python_api.drone_interface_base import DroneInterfaceBase
ModuleNotFoundError: No module named 'as2_python_api'
```

pariaspe, 2023-09-27:
ROS packages are sourced from `.bashrc` which is meant to be used in interactive sessions only:

```
$ head ./bashrc
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
[ -z "$PS1" ] && return
```

Script is returning before sourcing when the entrypoint is run directly. Easiest way to solve is to overwrite `.bashrc` file. We can also create another script for sourcing or source it from `entrypoint.sh` directly (or `.env`). What do you think @pawanw17 ?

pawanw17, 2023-09-27:
Having the sources in .bashrc certainly helps during debugging and development, as we don't have to source them every time we open a new terminal attached to docker. I was wondering if there is a file that is sourced in both interactive and non-interactive shells.

I am not sure maybe overwriting .bashrc could lead to other errors,  not too confident of this solution.

I'd suggest that we have it in .bashrc as well as .env/entrypoint.sh

pariaspe, 2023-09-27:
Ok, then I'll create a `.bash_profile` with all needed sources and this file will be source both in `.bashrc` and `entrypoint.sh`. 

Also, I've been reading `bash` manual entry and I might come with another solution:

```
$ man bash

       ...

       The  following paragraphs describe how bash executes its startup files.
       If any of the files exist but cannot be read, bash  reports  an  error.
       Tildes  are expanded in filenames as described below under Tilde Expan‐
       sion in the EXPANSION section.

       When bash is invoked as an interactive login shell, or as a  non-inter‐
       active  shell with the --login option, it first reads and executes com‐
       mands from the file /etc/profile, if that file exists.   After  reading
       that file, it looks for ~/.bash_profile, ~/.bash_login, and ~/.profile,
       in that order, and reads and executes commands from the first one  that
       exists  and  is  readable.  The --noprofile option may be used when the
       shell is started to inhibit this behavior.

       When an interactive login shell exits, or a non-interactive login shell
       executes  the  exit  builtin  command, bash reads and executes commands
       from the file ~/.bash_logout, if it exists.

       When an interactive shell that is not a login shell  is  started,  bash
       reads  and  executes  commands  from /etc/bash.bashrc and ~/.bashrc, if
       these files exist.  This may be inhibited by using the  --norc  option.
       The  --rcfile  file option will force bash to read and execute commands
       from file instead of /etc/bash.bashrc and ~/.bashrc.

       When bash is started non-interactively, to run a shell script, for  ex‐
       ample,  it  looks for the variable BASH_ENV in the environment, expands
       its value if it appears there, and uses the expanded value as the  name
       of  a  file to read and execute.  Bash behaves as if the following com‐
       mand were executed:
              if [ -n "$BASH_ENV" ]; then . "$BASH_ENV"; fi
       but the value of the PATH variable is not used to search for the  file‐
       name.

```

pawanw17, 2023-09-28:
yes, .bash_profile that should be good, can you please also update the dockerfiles at places where we write to bashrc

pawanw17, 2023-09-28:
@pariaspe I tested adding stuff in ~/.bash_profile and ~/.profile and opened a new terminal using docker exec

and ran ros2 list and got errors indicating the files weren't sourced, this is probably due because our shells aren't executed with --login

TL;DR ~/.bash_profile won't work for us

pariaspe, 2023-09-28:
Many thanks for the testing @pawanw17 :+1: 

I figured out a smoothly way to do it. It works for non-interactive and interactive shells. Setting BASH_ENV to the file where we are setting up the environment: `.env`

closed 2023-10-03 by pariaspe