JdeRobot/RoboticsAcademy #2299: [Obstacle Avoidance] improvements in the forces GUI display

jmplaza, 2023-10-23:
- [ ] the car should not be red, as the repulsive forces are already red, and so they are not distinguished. Maybe gold+black, as the current F1 car.
- [ ] the car should be located in the center of the display, not in the lower part. This way the repulsive forces will be properly displayed, they usually point to the rear part of the car as the laser sensor is looking forward
- [ ] 10m and 25m are wrong, they should be replaced by real distances in meters. They can be estimated according to the laser sensor readings.

AnushkaGharage, 2024-01-27:
1. I have changed the color to gold
2. Located the car in the center

I didn't quite understand the third issue. Will you please elaborate on that more?

jmplaza, 2024-01-27:
Sure @AnushkaGharage ,

the 10m and 25m rectangles at the current frontend are not located in true real distances.
[image] They should be replaced with the real distances, and maybe moved a little bit to "round numbers".

Could you share a snapshot of the frontend with your changes?

AnushkaGharage, 2024-01-29:
@jmplaza I tried running 'yarn run dev' and it shows following error -
"""
ERROR in ./src/exercise-index.js 91:19-62
Module not found: Error: Can't resolve 'exercises' in 'C:\Users\ghara\OneDrive\Desktop\RADI\RoboticsAcademy\react_frontend\src'
"""
WebpackError at /exercises/
            ModuleNotFoundError in 
            Module not found: Error: Can't resolve 'exercises' in 'C:\Users\ghara\OneDrive\Desktop\RADI\RoboticsAcademy\react_frontend\src'
"""
But there is no directory inside src named exercises
So I am trying to find a solution for this error.
I will definitely get back once I solve the error.

damodardatta, 2024-03-29:
[image]
Hey @jmplaza,
        I have changed the color of the position of arrows and laser scans, but i'm not sure about how to move the rectangles(10m and 25m) or convert their scale.

jmplaza, 2024-03-31:
Well done @damodardatta. The rectangles are wrong now, their motivation is to show to the student an estimate of the real distances around the car. Maybe 10m and 25m are too far, and the good numbers should be 3m and 5m.  Maybe you can look at the laser values (as they are measured in meters or centimeters) to set the rectangle scale...

damodardatta, 2024-04-02:
[image]

Hi @jmplaza, i changed the values and changed the rectangles to semicircle as i felt they would be more intuitive. Let know if this is ok?

javizqh, 2024-07-03:
As this UI has already been updated I will close this issue.
[image]

If you want to continue modifying this UI please open a new one

closed 2024-07-03 by javizqh