JdeRobot/RoboticsAcademy #2302: [Drone Rescue People] The drone takes off but does not move further

jmplaza, 2023-10-31:
This code fails, and it should work. Maybe problems inside `get_landed_state`

```
from GUI import GUI
from HAL import HAL
while True:
    if(HAL.get_landed_state()<=1):#take off if landed
      HAL.takeoff(5)
    else:#move if flying
      HAL.set_cmd_pos(20, 20, 5, 0)
```

This problem appears with the brand new release of the exercise in ROS2 Humble (RADI 4.4.16).

pawanw17, 2023-11-03:
I was able to reproduce the error, the issue is because of yaw angle being 0

I ran HAL.set_cmd_pos(20, 20, 5, 0.6) which worked fine,

pawanw17, 2023-11-03:
The issue arises from here https://github.com/aerostack2/aerostack2/blob/7e20ecdd03e71c6c0be83a196bc583ed6b003e5a/as2_motion_reference_handlers/as2_motion_reference_handlers/position_motion.py#L99

@pariaspe we can have some typecasting before calling aerostack to fix this

closed 2023-11-04 by pariaspe