JdeRobot/RoboticsAcademy #2303: [Humble] Exercise didn't terminate correctly

pawanw17, 2023-11-03:
While testing Rescue People exercise in Humble, the exercise loop got stuck and kept running even after exiting to menu.
[image]

pariaspe, 2023-11-07:
Hey @ango1994 , could you take a look at this? Thanks!

ango1994, 2023-11-09:
Hi @pawanw17! Could you provide more information about this issue? I just launched the RADI 4.4.19 and executed some code to make the drone take off. After closing every process terminates correctly. How can I replicate this error?

pawanw17, 2023-11-10:
@ango1994, Unfortunately, this is an intermittent issue and I do not have specific steps to reproduce it, I have come across this a few times on the latest RADI (4.4.19).

pawanw17, 2024-01-03:
Hi @ango1994 , fortunately I found a way to reproduce this issue try running this code in the editor and then restart and play the exercise, this crashes the exercise

```python
from GUI import GUI
from HAL import HAL
import time

print("Hi")
time.sleep(100)

while True:
    # Enter iterative code!
    print("Hello")
```

pawanw17, 2024-04-22:
Solved in superthin refactoring #

closed 2024-04-22 by pawanw17