JdeRobot/RoboticsAcademy #2315: Cannot import epylint form pylint

VlasanRobert, 2023-11-11:
Hello 

I have an issue when i try to set up the environment i get the followingerror when i want to run the python3 manage.py runserver
ImportError: cannot import name 'epylint' from 'pylint' (/home/robiti/Documents/RoboticsAcademy/env/lib/python3.11/site-packages/pylint/__init__.py)

pawanw17, 2023-11-12:
The issue is because epylint was removed from pylint in v3.0.0

pawanw17, 2023-11-12:
As a workaround, uninstall pylint and instead install this older version.

`pip3 install pylint==2.17.4`

VlasanRobert, 2023-11-12:
thank you for your help 
i still have one issue when i run the dokcer i get the following error:docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.

jmplaza, 2023-11-12:
Take a look at this answer [https://github.com/JdeRobot/RoboticsAcademy/issues/2317#issuecomment-1807154635](https://github.com/JdeRobot/RoboticsAcademy/issues/2317#issuecomment-1807154635)

PR #2316 (2023-11-12): Fixed pylint version and fixed runserver command
Fixes #2315 

By default runserver runs on port 8000 and we have migrated to port 7164.
closed 2023-11-12 by pariaspe