JdeRobot/RoboticsAcademy #2332: Dockerfile dependencies humble build fails

ReyDoran, 2023-11-28:
Error:
```Step 47/51 : RUN /bin/bash -c "source /opt/ros/humble/setup.bash; colcon build --symlink-install"
Step 47/51 : RUN /bin/bash -c "source /opt/ros/humble/setup.bash; colcon build --symlink-install"
 ---> Running in 1591e0e5c50f
Starting >>> as2_msgs
Starting >>> px4_msgs
Starting >>> as2_cli
Starting >>> as2_gazebo_classic_assets
Finished <<< as2_cli [1.68s]
Finished <<< as2_gazebo_classic_assets [2.20s]
[Processing: as2_msgs, px4_msgs]
[Processing: as2_msgs, px4_msgs]
Finished <<< as2_msgs [1min 26s]
Starting >>> as2_core
--- stderr: as2_core
In file included from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/sensor.hpp:63,
                 from /home/drones_ws/src/aerostack2/as2_core/src/sensor.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_core/src/utils/tf_utils.cpp:33:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/utils/frame_utils.hpp:38,
                 from /home/drones_ws/src/aerostack2/as2_core/src/utils/frame_utils.cpp:34:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/utils/frame_utils.hpp:38,
                 from /home/drones_ws/src/aerostack2/as2_core/tests/frame_test.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/src/aerostack2/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_core/tests/tf2_namespace_test.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
---
[Processing: as2_core, px4_msgs]
[Processing: as2_core, px4_msgs]
Finished <<< as2_core [1min 15s]
Starting >>> as2_motion_reference_handlers
Starting >>> as2_behavior
Starting >>> as2_alphanumeric_viewer
Starting >>> as2_state_estimator
[Processing: as2_alphanumeric_viewer, as2_behavior, as2_motion_reference_handlers, as2_state_estimator, px4_msgs]
Finished <<< as2_behavior [36.2s]
--- stderr: as2_motion_reference_handlers
In file included from /home/drones_ws/install/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/include/as2_motion_reference_handlers/speed_in_a_plane_motion.hpp:40,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/src/speed_in_a_plane_motion.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/install/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/include/as2_motion_reference_handlers/speed_motion.hpp:40,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/src/speed_motion.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/drones_ws/install/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/include/as2_motion_reference_handlers/position_motion.hpp:40,
                 from /home/drones_ws/src/aerostack2/as2_motion_reference_handlers/src/position_motion.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
[... 62 lines cut ...]
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< as2_alphanumeric_viewer [1min 26s]
--- stderr: as2_motion_controller
CMake Error: install(EXPORT "export_as2_motion_controller" ...) includes target "as2_motion_controller" which requires target "pid_controller" that is not in any export set.
CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< as2_motion_controller [14.1s, exited with code 1]
Aborted  <<< as2_state_estimator [2min 16s]
Aborted  <<< as2_behaviors_motion [1min 54s]
Aborted  <<< px4_msgs [6min 41s]

Summary: 9 packages finished [6min 43s]
  1 package failed: as2_motion_controller
  3 packages aborted: as2_behaviors_motion as2_state_estimator px4_msgs
  7 packages had stderr output: as2_behaviors_motion as2_core as2_keyboard_teleoperation as2_motion_controller as2_motion_reference_handlers as2_python_api as2_state_estimator
  3 packages not processed
The command '/bin/sh -c /bin/bash -c "source /opt/ros/humble/setup.bash; colcon build --symlink-install"' returned a non-zero code: 1
```

pariaspe, 2023-11-28:
Hey @ReyDoran , a dependency of `as2_motion_controller` has changed. I will open a PR fixing this soon.

pariaspe, 2023-11-28:
Building should be back to work @ReyDoran. If it is still failing, feel free to ping me

closed 2023-11-28 by pariaspe