JdeRobot/RoboticsAcademy #2341: Add new Worlds model to database

ReyDoran, 2023-12-01:
World information will be stored in a new model "World" containing launch file location or launch file contents, resource folders, model folders, ROS version and name

pariaspe, 2023-12-01:
Hi @ReyDoran , why to use a database entry while the standard way of doing this is by environment variables (`GAZEBO_RESOURCE_PATH`, `GAZEBO_MODEL_PATH`, etc)? 

We should set the variables while building a ROS pkg, so it could be avoided set them on the RAM launching. @pawanw17 worked on this in this [PR](https://github.com/JdeRobot/RoboticsApplicationManager/pull/55/files).

What do you think @jmplaza @ango1994 ?

jmplaza, 2023-12-02:
We recently moved the world files from each exercise directory to RoboticsInfrastructure repo, so they could be used not only from RoboticsAcademy but from other applications such VisualCircuit, bt-studio... as well. Just to increase interoperability.

Typically we had for each exercise a single world file, and the collection of available worlds was not organized at all. Each world was strongly linked to the single exercise that used it. In addition, now some exercises accept several "scenarios", each scenario is a different launcher file (which includes a different Gazebo world). 

The point of this issue is to create a "World DB" to store all the possible launcher files available for RoboticsAcademy, VisualCircuit, bt-studio.... And in the "Exercise DB" inside RoboticsAcademy we could store the relevant information about each exercise, such as its state, the possible 'worlds' for it, etc...

All the model folders and environment variables (GAZEBO_RESOURCE_PATH, GAZEBO_MODEL_PATH...) should be properly set inside the RADI so the gzserver always finds everything it needs when it is launched. That information should not be "exercise specific", as all the world files are now stored at RoboticsInfrastructure repository.

pariaspe, 2023-12-03:
> All the model folders and environment variables (GAZEBO_RESOURCE_PATH, GAZEBO_MODEL_PATH...) should be properly set inside the RADI so the gzserver always finds everything it needs when it is [...]

That's what I tried to said in my previous message. I would recommend to create a ROS pkg and hook all world, models or other assets on the CMakeLists.txt, as we are doing on the [ROS2 RADI](https://github.com/JdeRobot/RoboticsInfrastructure/blob/humble-devel/CustomRobots/CMakeLists.txt).

closed 2023-12-14 by ReyDoran