JdeRobot/RoboticsAcademy #2356: GZViewer not launching with Nvidia GPU Acceleration

ReyDoran, 2023-12-11:


jmplaza, 2024-02-20:
Gzviewer is launching and also the console window on the correspoding vnc DISPLAYs, but those displays show black screens. We have a problem with Nvidia GPU acceleration in ROS2 Humble RADIs. They work fine on ROS1-Noetic RADIs though.

CDAM2020, 2024-07-16:
When the GPU is integrated on CPU It doesn't appear

dduro2020, 2024-07-23:
I'm having the same issue, but I don't know what could have caused it, it used to work without any problems.

jmplaza, 2024-07-23:
With RoboticsBackend 4.6.3 @dduro2020 ?

dduro2020, 2024-07-23:
Yes, when I updated my fork to the latest commits it started to crash.

dduro2020, 2024-07-23:
I noticed that the browser buttons don't update (the console does), you can launch the code (without seeing what's going on), but you can't stop it or restart it

CDAM2020, 2024-07-23:
Can you specify the steps to reproduce this error? I've tested on some exercises and this error didn't appear.

dduro2020, 2024-07-23:
I was just launching the script `sh scripts/develop_academy.sh -g` and opening the exercises, for some reason it solved itself, I attach a picture of how I saw the error (I think the problem was that some browser components were frozen)
[image]

jmplaza, 2024-07-23:
That is an instruction for developers, and I'm not sure whether it is outdated or not. Please check this issue as a regular user, just launching properly the RoboticsBackend and using RoboticsAcademy from the browser.

dduro2020, 2024-07-23:
I've been trying it both ways and got the same result, but I mean that by launching it this way it has been magically fixed

dduro2020, 2024-07-24:
After verifying that I was launching the wrong user version, I close the issue to open another one specifically for launching it for developers

closed 2024-07-24 by dduro2020

jmplaza, 2024-07-24:
Glad to see that following the launching instructions at the Robotics Academy [User Guide ](https://jderobot.github.io/RoboticsAcademy/user_guide/) and using the latest RoboticsBackend (4.6.3) this problem is completeley solved!