JdeRobot/RoboticsAcademy #2382: [MonteCarlo Laser Loc] introduce several optimization techniques in the user doc

jmplaza, 2023-12-20:
that can be used. For instance downsampling the laser readings (180 ---> 18?) and computing the theoretical laser readings raytracing on a smaller map image (1024x1024 -----> 400 x 400?). 

Applying them the number of particles on real-time operation of the localization algorithm can be larger, and so the relocalization power of the MCL algorithm.

vinayakgavariya, 2023-12-27:
hey @jmplaza i would like to work on this issue.

can you please assign this to me ?

also it will be helpful if you can send the required links for this issue i.e.. user doc

jmplaza, 2023-12-27:
uhm... @franmore-urjc is working on this issue. He has used this exercise on his classes and knows the most relevant techniques to optimize the solution, including multiprocessing module.  So I suggest you focus on a different issue.

Cheers

vinayakgavariya, 2023-12-28:
ahh no worries, hope it resolves soon.

actually i am new to JdeRobot and RoboticsAcademy as well so will you suggest few things where i can start contributing?
or if you have something where i can help you with please let me know.

i am familiar with python and currently practising machine learning.

jmplaza, 2023-12-28:
Sure.

A good way to start contributing is to get familiar with our software products. In particular, RoboticsAcademy is the most active one and a priority project. The best way to start contributing is to **USE** RoboticsAcademy (solve and provide feedback) about the exercises already available in current release (FollowLine, ObstacleAvoidance, VacuumCleaner, LocalizedVacuumCleaner...). They use the browser as the only GUI (for source code editing and for exercise monitoring), and a docker image for running the Gazebo simulator (it is named RADI= RoboticsAcademy Docker Image). The docker image already includes a Django webserver for providing the exercise webpages. Please use the [RoboticsAcademy GitHub Discussions](https://github.com/JdeRobot/RoboticsAcademy/discussions) for questions (you will receive support there) and report any detected bug creating an issue for each one in the RoboticsAcademy repository. Feel free to share videos in the [Unibotics forum](https://forum.unibotics.org/) or mentioning @JdeRobot in tweets of your solutions to those exercises in operation. 

After using RoboticsAcademy, a second step in contributing is studying its source code and **improve** it.

jmplaza, 2025-11-09:
Hi @Sarvesh-Mishra1981 , 

@Butakus has already several solutions already developed for this issue AFAIK (the PR is pending). I suggest you work on betatesting the Car Junction exercise (#3231 and #3256 ). Your contributions there would be very useful. What do you think?