JdeRobot/RoboticsAcademy #2389: [New Commers Issue] script deprecation warning

pawanw17, 2024-01-02:
There is a warning in an onld script that we use that leads to 1 min wait time, replace this script with ideal updated one

`RUN curl -sL https://deb.nodesource.com/setup_16.x | bash -   && apt-get install -y nodejs   && npm install -g yarn `

[image]

lavish123409, 2024-01-14:
I would like to work on this issue. Just to ask one time, is this issue resolved or anyone else is working on this issue?

pawanw17, 2024-01-14:
@lavish123409 Feel free to work on this issue, you can find the steps to build a RADI here: https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/generate_a_mini_radi.md

lavish123409, 2024-01-14:
So according to my understanding, `RUN curl -sL https://deb.nodesource.com/setup_16.x | bash -   && apt-get install -y nodejs   && npm install -g yarn` is the old script for which I have to search a new ideal one and then update it. Am I going right?

pawanw17, 2024-01-15:
@lavish123409 Yes that's correct

lavish123409, 2024-01-17:
@pawanw17 On digging up on the problem, I found out that it is trying to install the nodejJS version which is deprecated. So, I should change the version of node to either 21x, 20x or 18x as mentioned [here](https://github.com/nodesource/distributions#installation-instructions). So, should I choose any specific version that does not break any code or we can use any node version?

pawanw17, 2024-01-21:
Hi @lavish123409, we would prefer the latest version on node that doesn't break our code, I see that you have already done that, that's great! Can you also update the dependencies-humble docker file
https://github.com/JdeRobot/RoboticsAcademy/blob/74e2848f9d2164b3ed65387348c834a683c9ad4f/scripts/mini_RADI/Dockerfile.dependencies_humble#L135

Make sure to locally test the UI

lavish123409, 2024-01-28:
Hey @pawanw17, I have made that change in the file. https://github.com/JdeRobot/RoboticsAcademy/blob/74e2848f9d2164b3ed65387348c834a683c9ad4f/scripts/mini_RADI/Dockerfile.dependencies_humble#L135

Can you tell me how can I locally test the UI? Thanks

jmplaza, 2024-01-28:
Hi @lavish123409 ,

take a look at the [documentation for developers of RoboticsAcademy](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md). 

The CustomRobots repository has been renamed as[ RoboticsInfrastructure ](https://github.com/JdeRobot/RoboticsInfrastructure), which has now a CustomRobots directory. There are some updates to the documentation pending, but it is a good reference.

lavish123409, 2024-01-28:
It failed on my system. Don't know whether it is because of my change or some other problem. Here, is the error.
```bash
=> ERROR [10/26] RUN /bin/bash -c '. /opt/ros/$ROS_DISTRO/setup.bash; cd /catkin_ws; catkin build'                                                                                                  540.6s 
------                                                                                                                                                                                                      
 > [10/26] RUN /bin/bash -c '. /opt/ros/$ROS_DISTRO/setup.bash; cd /catkin_ws; catkin build':                                                                                                               
3.108 -----------------------------------------------                                                                                                                                                       
3.108 Profile:                     default                                                                                                                                                                  
3.108 Extending:             [env] /opt/ros/noetic                                                                                                                                                          
3.108 Workspace:                   /catkin_ws                                                                                                                                                               
3.108 -----------------------------------------------
3.108 Build Space:        [exists] /catkin_ws/build
3.108 Devel Space:        [exists] /catkin_ws/devel
3.108 Install Space:      [unused] /catkin_ws/install
3.108 Log Space:         [missing] /catkin_ws/logs
3.108 Source Space:       [exists] /catkin_ws/src
3.108 DESTDIR:            [unused] None
3.108 -----------------------------------------------
3.108 Devel Space Layout:          linked
3.108 Install Space Layout:        None
3.108 -----------------------------------------------
3.108 Additional CMake Args:       None
3.108 Additional Make Args:        None
3.108 Additional catkin Make Args: None
3.108 Internal Make Job Server:    True
3.108 Cache Job Environments:      False
3.108 -----------------------------------------------
3.108 Buildlisted Packages:        None
3.108 Skiplisted Packages:         None
3.108 -----------------------------------------------
3.108 Workspace configuration appears valid.
3.108 
3.108 NOTE: Forcing CMake to run for each package.
3.108 -----------------------------------------------
3.108 [build] Found 24 packages in 0.0 seconds.                                      
3.108 [build] Updating package table.                                                
3.108 Starting >>> catkin_tools_prebuild                                             
Finished <<< catkin_tools_prebuild                  [ 2.5 seconds ]            ld... 
5.621 Starting >>> drone_assets                                                      
5.621 Starting >>> drone_circuit_assets                                              
5.621 Starting >>> kobuki_gazebo_plugins                                             
5.621 Starting >>> mavlink                                                           
5.621 Starting >>> mavros_msgs                                                       
5.621 Starting >>> rqt_ground_robot_teleop                                           
5.621 Starting >>> stdr_msgs                                                         
5.621 Starting >>> stdr_resources                                                    
Finished <<< stdr_resources                         [ 4.0 seconds ]             0.1] 
Finished <<< drone_circuit_assets                   [ 4.1 seconds ]            4.... 
Finished <<< rqt_ground_robot_teleop                [ 5.2 seconds ]            in... 
Finished <<< mavlink                                [ 5.7 seconds ]            in... 
Finished <<< stdr_msgs                              [ 15.5 seconds ]           in... 
21.04 Starting >>> libmavconn                                                        
Finished <<< mavros_msgs                            [ 1 minute and 44.3 seconds ]... 
Finished <<< libmavconn                             [ 5 minutes and 54.9 seconds ].. 
Failed <<< drone_assets                           [ 6 minutes and 34.5 seconds ]e... 
400.6 Abandoned <<< mavros                                 [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_parser                            [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_samples                           [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_server                            [ Unrelated job failed ]  
400.6 Abandoned <<< jderobot_drones                        [ Unrelated job failed ]  
400.6 Abandoned <<< drone_wrapper                          [ Unrelated job failed ]  
400.6 Abandoned <<< mavros_extras                          [ Unrelated job failed ]  
400.6 Abandoned <<< rotors_driver                          [ Unrelated job failed ]  
400.6 Abandoned <<< rqt_drone_teleop                       [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_launchers                         [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_robot                             [ Unrelated job failed ]  
400.6 Abandoned <<< stdr_gui                               [ Unrelated job failed ]  
[... cut ...]
Dockerfile.mini_noetic:29
--------------------
  27 |     
  28 |     RUN cd /catkin_ws && rosdep update && rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO -y
  29 | >>> RUN /bin/bash -c '. /opt/ros/$ROS_DISTRO/setup.bash; cd /catkin_ws; catkin build'
  30 |     
  31 |     #ROScontrol and LIDAR
--------------------
ERROR: failed to solve: process "/bin/sh -c /bin/bash -c '. /opt/ros/$ROS_DISTRO/setup.bash; cd /catkin_ws; catkin build'" did not complete successfully: exit code: 1

```

pawanw17, 2024-02-01:
@lavish123409 Unfortunately I am not seeing this issue while building the RADI, I believe you used parameters of humble in the build.sh script while building noetic.

To build for noetic please use something like
./build.sh -f --academy master --infra noetic-devel --ram main --ros noetic --tag noetic_master

pawanw17, 2024-02-01:
It seems that our code breaks on the latest script

lavish123409, 2024-02-01:
Hey @pawanw17, I think that the UI is working fine.

[image]
I am supposing that I would not be able to open any exercise without the RADI.
[image]

Regarding this comment, Is there anything that I can do on my side?

pawanw17, 2024-02-02:
Hi @lavish123409 I have checked again by building the RADI, the version seems to break the UI, you might be using an old version on your machine while testing the UI, please confirm the output of 

```
node -v
``` 

[image]

lavish123409, 2024-02-02:
Yes @pawanw17, indeed it is an older version of node.
```bash
$ node -v
v12.22.9
```

What version of node should I upgrade to in order to see the error or should I just change the version of node in the Dockerfile, so that the error resolves?

pawanw17, 2024-02-02:
Removing and Installing different node versions on your primary machine could mess things up for you. I suggest changing the node version in a docker image and try building our UI and see which version doesn't break it. The steps for building the UI can be found here: https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md

lavish123409, 2024-02-03:
I ran the build script again with node 20x in Dockerfile but I faced same error again. On seeing this [error](https://github.com/JdeRobot/RoboticsAcademy/issues/2389#issuecomment-1922870427), I realised that it is a different error from mine and it is happening after that command which is giving my error.

I am facing error at this command in Dockerfile
```bash
RUN /bin/bash -c '. /opt/ros/$ROS_DISTRO/setup.bash; cd /catkin_ws; catkin build'
```
and I don't have any idea about what does this command do and how to resolve it?
Any suggestions @pawanw17

pawanw17, 2024-02-03:
@lavish123409 How are you running the script can you share the command?

lavish123409, 2024-02-03:
Just the `./build.sh` command
```bash
$ ./build.sh
ROBOTICS_ACADEMY:-------------:master
ROBOTICS_INFRASTRUCTURE:------:noetic-devel
RAM:--------------------------:main
ROS_DISTRO:-------------------:noetic
IMAGE_TAG:--------------------:test

===================== BUILDING noetic BASE IMAGE =====================
Building base using Dockerfile.dependencies_noetic for ROS noetic
[+] Building 85.7s (34/34) FINISHED  
```

pawanw17, 2024-02-03:
@lavish123409 I have a hunch that this is because something went wrong while building the base image can you try

```
./build.sh -F
``` 

This will force rebuilding of the base RADI

lavish123409, 2024-02-03:
The output of command is this.
```bash
$ ./build.sh
ROBOTICS_ACADEMY:-------------:master
ROBOTICS_INFRASTRUCTURE:------:noetic-devel
RAM:--------------------------:main
ROS_DISTRO:-------------------:noetic
IMAGE_TAG:--------------------:test

===================== BUILDING noetic BASE IMAGE =====================
Building base using Dockerfile.dependencies_noetic for ROS noetic
[+] Building 85.7s (34/34) FINISHED                                                                                                                                                    docker:desktop-linux
 => [internal] load build definition from Dockerfile.dependencies_noetic                                                                                                                               0.5s
 => => transferring dockerfile: 10.76kB                                                                                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                                                                      0.4s
 => => transferring context: 2B                                                                                                                                                                        0.1s
 => [internal] load metadata for docker.io/nvidia/opengl:1.2-glvnd-runtime-ubuntu20.04                                                                                                                 3.9s
 => [ 1/30] FROM docker.io/nvidia/opengl:1.2-glvnd-runtime-ubuntu20.04@sha256:4480cecadba0bd3763d7c4934225620210527e3d985dff0e34855cb87e2fc1db                                                         0.0s
 => CACHED [ 2/30] RUN apt-get clean &&     apt-get update &&     apt-get install --no-install-recommends -y locales &&     rm -rf /var/lib/apt/lists/* &&     locale-gen en_US.UTF-8                  0.0s
 => CACHED [ 3/30] RUN apt-get update && apt-get install -y         software-properties-common         apt-utils         build-essential make gcc git         ca-certificates         curl wget        0.0s
 => CACHED [ 4/30] RUN tasksel install lubuntu-core lubuntu-desktop                                                                                                                                    0.0s
 => CACHED [ 5/30] RUN wget --no-check-certificate https://deac-fra.dl.sourceforge.net/project/virtualgl/3.0/virtualgl_3.0_amd64.deb &&     wget --no-check-certificate https://kumisystems.dl.source  0.0s
 => CACHED [ 6/30] RUN git clone https://github.com/novnc/noVNC.git -b v1.2.0 &&  cd noVNC/utils &&  git clone https://github.com/novnc/websockify websockify -b v0.9.0                                0.0s
 => CACHED [ 7/30] RUN apt-get update && apt-get install -y --no-install-recommends         dbus-x11         libdbus-c++-1-0v5 &&     rm -rf /var/lib/apt/lists/*                                      0.0s
 => CACHED [ 8/30] RUN apt-get update && apt-get -y --quiet --no-install-recommends install         bzip2         ca-certificates         ccache         cmake         cppcheck         curl           0.0s
 => CACHED [ 9/30] RUN apt-get update && apt-get install -y libgtest-dev cmake &&     cd /usr/src/gtest  && mkdir build && cd build  && cmake .. && make -j$(nproc)  && find . -name *.a -exec cp {}   0.0s
 => CACHED [10/30] RUN python3 -m pip install --upgrade pip==23.2.1 wheel==0.41.2 setuptools==68.2.2                                                                                                   0.0s
 => CACHED [11/30] RUN python3 -m pip install argparse==1.4.0 argcomplete==3.1.2 coverage==7.3.2 cerberus==1.3.5 empy==3.3.4 jinja2==3.1.2 kconfiglib==14.1.0   matplotlib==3.0.* numpy nunavut==1.1.  0.0s
 => CACHED [12/30] RUN ln -sf /usr/bin/ccache /usr/lib/ccache/cc  && ln -sf /usr/bin/ccache /usr/lib/ccache/c++                                                                                        0.0s
 => CACHED [13/30] RUN wget -q https://downloads.sourceforge.net/project/astyle/astyle/astyle%203.1/astyle_3.1_linux.tar.gz -O /tmp/astyle.tar.gz  && cd /tmp && tar zxf astyle.tar.gz && cd astyle/s  0.0s
 => CACHED [14/30] RUN wget -q "https://services.gradle.org/distributions/gradle-6.3-rc-4-bin.zip" -O /tmp/gradle-6.3-rc-4-bin.zip  && mkdir /opt/gradle  && cd /tmp  && unzip -d /opt/gradle gradle-  0.0s
 => CACHED [15/30] RUN git clone https://github.com/eProsima/foonathan_memory_vendor.git /tmp/foonathan_memory  && cd /tmp/foonathan_memory  && mkdir build && cd build  && cmake ..  && cmake --buil  0.0s
 => CACHED [16/30] RUN git clone --recursive https://github.com/eProsima/Fast-DDS.git -b v2.1.0 /tmp/FastRTPS-2.1.0  && cd /tmp/FastRTPS-2.1.0  && mkdir build && cd build  && cmake -DTHIRDPARTY=ON   0.0s
 => CACHED [17/30] RUN git clone --recursive https://github.com/eProsima/Fast-DDS-Gen.git -b v1.0.4 /tmp/Fast-RTPS-Gen-1.0.4  && cd /tmp/Fast-RTPS-Gen-1.0.4  && gradle assemble  && gradle install    0.0s
 => CACHED [18/30] RUN wget --quiet http://packages.osrfoundation.org/gazebo.key -O - | apt-key add -  && sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -sc` m  0.0s
 => CACHED [19/30] RUN apt-get update && apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654  && sh -c 'echo "deb http://packages.ros.org/ros/u  0.0s
[... cut ...]
What's Next?                                                                                                                                                                                                
  1. Sign in to your Docker account → docker login
  2. View a summary of image vulnerabilities and recommendations → docker scout quickview
```
So, I think the base image is getting created without any errors.
Basically, I deleted the created image in docker-desktop and then running this command, so it is building the base image first, I believe.
Also, I wanted to ask that is there any chat channel for conversation as I am thinking this issue thread is getting too long. :sweat_smile:

pawanw17, 2024-02-03:
@lavish123409, Did the above command also result in an error? it is supposed to build the noetic RADI after the base image

We do not have a chat channel, However you can always initiate discussions here https://github.com/orgs/JdeRobot/discussions

lavish123409, 2024-02-04:
Yes @pawanw17, `./build.sh -F` command also resulted in error at same point like in [this](https://github.com/JdeRobot/RoboticsAcademy/issues/2389#issuecomment-1925181946) error. Although, the base image is being able to build compeletly without errors.

lavish123409, 2024-02-12:
I am continuously receiving this error of not able to connect to deac-fra.dl.sourceforge.net because of which the build of mini RADI image is failing.

[image]

Any suggestions on how to resolve that?

lavish123409, 2024-02-18:
Hey @pawanw17 @jmplaza, I am still receiving this error and after searching on the internet a lot, I think maybe it is because of shortage of disk space.

[image]

Can anyone tell me how much space would the RADI image require, so that I can allocate that much space in docker desktop?
Thanks

pawanw17, 2024-02-18:
Hi @lavish123409, I am not able to reproduce your error. I use docker CLI and AFAIK space is allocated dynamically from the system as per the image's requirements.

jmplaza, 2024-02-18:
uhm... @lavish123409 , RoboticsAcademy requires some HardDrive space to store the RADI and some RAM memory space while running. 

Regarding hard drive space, running "docker images" at my Linux computer tells these sizes for the RADIs
```
REPOSITORY                  TAG                           IMAGE ID       CREATED         SIZE
jderobot/robotics-academy   3.4.29                        68a606c2f52a   9 days ago      12GB
jderobot/robotics-academy   4.4.27                        3d7263a8d639   2 months ago    14.7GB
```
At [RADI DockerHub](https://hub.docker.com/r/jderobot/robotics-academy/tags) they are compressed. Some illustrative sizes:
RADI 4.4.32, 6.07 GB
RADI 3.4.30, 4.77 GB

On execution I successfully run RoboticsAcademy on my computer with 32GB, and I know of successful executions in 8GB RAM. Maybe 4GB are too short to be fluent...

pawanw17, 2024-02-18:
@lavish123409 , can you share more about your system's OS, RAM, cores etc. To help debug this behaviour?

jmplaza, 2024-02-18:
@lavish123409 RoboticsAcademy should work like a charm on that machine. Have you _used_ it? (not with the developer deployment, just as a regular user: 

1. download the RADI from DockerHub. For instance 3.4.30, with `docker pull jderobot/robotics-academy:3.4.30`
2. launch the RADI. For instance with `docker run --rm -it -p 7164:7164 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 -p 7163:7163 jderobot/robotics-academy:3.4.30`
3. start a browser and access to the https://127.0.0.1:7164/

lavish123409, 2024-02-18:
Ok @jmplaza, I would definitely try that but currently I need to see whether my changes in the script files are not breaking the code. Its been a long time working on this issue, and if that error can be resolved then we can wrap up this issue soon.

jmplaza, 2024-02-18:
@lavish123409  Once you are sure that everything works as user, that provides information about what could be failing on your developer deployment. It will help to discard possible sources of failure.

lavish123409, 2024-02-19:
Yes @jmplaza, the RA image seems to be working fine on my machine.

[image]

If any kind of problem is there or I need to check something in order to deduce that the RADI image is running properly, then kindly notify me. Thanks.

jmplaza, 2024-02-19:
Try with the FollowLine exercise in operation, not only the exercises grid. For instance with this dummy source code:
```
from GUI import GUI
from HAL import HAL
# Enter sequential code!

angular_speed = 0.3
i = 0
while True:
    # Enter iterative code!
    img = HAL.getImage()
    GUI.showImage(img)
    HAL.setW(angular_speed)
    print(i)
    i=i+1
```

lavish123409, 2024-02-20:
I think it is working fine.

[image]

lavish123409, 2024-02-20:
I think `catkin build` command is needed to build the RADI image, at which my execution is failing. What does this command do?

lavish123409, 2024-02-21:
Any suggestions @jmplaza @pawanw17 ?

pawanw17, 2024-02-23:
Hi @lavish123409, the execution of the pulled image seems correct, I am still not sure why your build is failing. 

For the testing of this issue I'd recommend building of a ROS2 Humble RADI and ROS1 Noetic RADI I use the following commands to build those

./build.sh -f --academy master --infra noetic-devel --ram main --ros noetic --tag <TAG_NAME>

./build.sh -f --academy master --infra humble-devel --ram main --ros humble --tag <TAG_NAME>

The -f flag checks whether the base image exists or not and if it doesn't then builds it before building the main image. I also  add the -F flag when I feel the need to force the script to build the base image.

You can try building the humble image, and test that out, while I can test out the Noetic image and then if all goes well, we can close this issue.

lavish123409, 2024-02-25:
Hello @pawanw17, I am trying to run this command `./build.sh -f --academy master --infra humble-devel --ram main --ros humble --tag <TAG_NAME>` from yesterday. But, it is constantly giving this error message.

[image]

lavish123409, 2024-02-25:
Hey @pawanw17, I solved the `catkin build` error, but it has now stuck after running complete `Dockerfile`.

[image]

Is there anything that can be done here?

pawanw17, 2024-02-25:
@lavish123409, This is the last step of the RADI, it takes a few seconds and after that, the RADI is built

How did you solve the catikin issue?

lavish123409, 2024-02-25:
Yes, I solved that catkin issue. But are you sure that it takes some time.

>  This is the last step of the RADI, it takes a few seconds 

You can see in the image that it had already taken 10,000 seconds and I let it run for some more time, but the size of copied files was still 243.22MB.

pawanw17, 2024-02-25:
Something is not right, it should not take that long

lavish123409, 2024-02-25:
No, I don't think so, I am able to browse the internet normally.

I tried to find this error on internet, and found this [issue](https://github.com/moby/buildkit/issues/4453) related to our error which does not have any solution or workaround. Can you suggest anything for it?

lavish123409, 2024-02-26:
I have now changed the Docker version.
[image]

But, still the behaviour is same.
[image]

lavish123409, 2024-03-01:
The RADI image has been built but still I am not able to connect it to perform the exercises.
[image]
[image]

What can be the potential problem?

PR #2502 (2024-04-26): updated node version
Fixes #2389
closed 2024-04-26 by pawanw17