JdeRobot/RoboticsAcademy #2395: Unable to launch Exercises on M1 Mac

AryanNanda17, 2024-01-12:
Hi, I am unable to launch exercises on my M1 Mac:-
I followed [these](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md) instructions for the setup of RoboticsAcademy locally.
Whenever I run the docker image and try to launch an exercise, my screen gets stuck here:-
[image]
These are the logs generated on my terminal:-
```

2024-01-12 19:33:11,844 [Thread-2    ] [INFO ] (root)  client connected: {'id': 1, 'handler': <websocket_server.websocket_server.WebSocketHandler object at 0x4002755550>, 'address': ('192.168.65.1', 38279)}
2024-01-12 19:33:11,853 [Thread-2    ] [INFO ] (root)  message received: {"id":"cbd63e7c-4dd2-42a9-945e-bef32a73c760","command":"connect"} from client {'id': 1, 'handler': <websocket_server.websocket_server.WebSocketHandler object at 0x4002755550>, 'address': ('192.168.65.1', 38279)}
/dev/dri does not exist. No direct GPU access.
2024-01-12 19:33:12,055 [MainThread  ] [INFO ] (transitions.core)  Executed callback 'on_connect'
2024-01-12 19:33:12,056 [MainThread  ] [INFO ] (transitions.core)  Finished processing state idle exit callbacks.
2024-01-12 19:33:12,056 [MainThread  ] [INFO ] (root)  Connect state entered
2024-01-12 19:33:12,057 [MainThread  ] [INFO ] (transitions.core)  Executed callback 'on_enter_connected'
2024-01-12 19:33:12,057 [MainThread  ] [INFO ] (transitions.core)  Finished processing state connected enter callbacks.
2024-01-12 19:33:12,058 [MainThread  ] [INFO ] (root)  State changed to connected
2024-01-12 19:33:12,059 [MainThread  ] [INFO ] (transitions.core)  Executed callback '<bound method Manager.state_change of <__main__.Manager object at 0x40026f61c0>>'
2024-01-12 19:33:12,075 [Thread-2    ] [INFO ] (root)  message received: {"id":"10097cd5-4763-4fdc-8cf0-daa5f3ab462a","command":"launch","data":{"launch":{},"application":{"type":"python","params":{"circuit":"simple"},"entry_point":"$EXERCISE_FOLDER/entry_point/ros2_humble/exercise.py"},"exercise_id":"follow_line_newmanager","visualization":"gazebo_rae","world":"gazebo","resource_folders":"$EXERCISE_FOLDER","model_folders":"$CUSTOM_ROBOTS_FOLDER/f1/models","launch_file":"/opt/jderobot/Launchers/simple_circuit_followingcam.launch.py","name":"Simple"}} from client {'id': 1, 'handler': <websocket_server.websocket_server.WebSocketHandler object at 0x4002755550>, 'address': ('192.168.65.1', 38279)}
2024-01-12 19:33:12,162 [MainThread  ] [INFO ] (root)  Launch transition started, configuration: {'launch': {}, 'application': {'type': 'python', 'params': {'circuit': 'simple'}, 'entry_point': '$EXERCISE_FOLDER/entry_point/ros2_humble/exercise.py'}, 'exercise_id': 'follow_line_newmanager', 'visualization': 'gazebo_rae', 'world': 'gazebo', 'resource_folders': '$EXERCISE_FOLDER', 'model_folders': '$CUSTOM_ROBOTS_FOLDER/f1/models', 'launch_file': '/opt/jderobot/Launchers/simple_circuit_followingcam.launch.py', 'name': 'Simple', 'ros_version': '1'}
2024-01-12 19:33:13,487 [MainThread  ] [INFO ] (rospy.topics)  topicmanager initialized
Xserver on :0 is running!
... logging to /root/.ros/log/6d3eb760-b181-11ee-a7be-0242ac110002/roslaunch-8f3718ea9459-111.log
```

jmplaza, 2024-01-13:
Hi @AryanNanda17 ,

uhm... those instructions are for developer deployment of RoboticsAcademy. As regular user it is more simple:

1. Download the latest RADI (for ROS1-Noetic is 3.4.28) from [DockerHub](https://hub.docker.com/r/jderobot/robotics-academy/tags)
2. Start the RADI using: `docker run --rm -it -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 -p 7163:7163 -p 7164:7164 jderobot/robotics-academy:3.4.28`
3. Open your local browser and select this URL: `http://127.0.0.1:7164/`

AryanNanda17, 2024-01-13:
Thanks!!
I thought for contributing to RoboticsAcademy, I have to follow those instructions for setup and it kept giving me errors even after trying many different things. 
Anyways, what do you suggest from which exercise should I start?

jmplaza, 2024-01-13:
FollowLine and Basic Vacuum Cleaner as a regular user, try to solve them and enjoy :-). For studying the underlying code Vacuum Cleaner is the best one.

closed 2024-01-13 by jmplaza