JdeRobot/RoboticsAcademy #2418: [New Manager] Migrate all exercises

OscarMrZ, 2024-02-19:
# Noetic
- [ ] Follow line
- [ ] Basic Vacuum Cleaner
- [ ] Autoparking
- [ ] Localized Vacuum Cleaner
- [ ] Global Navigation
- [ ] Rescue people
- [ ] Obstacle Avoidance

# Humble
- [ ] Follow line
- [ ] Amazon Warehouse (Issue with rclpy)
- [ ] Drone Rescue People
- [ ] Autoparking (Issue with rclpy)

pawanw17, 2024-02-23:
Hi @Surajpatra700, Unfortunately this is more of an intermediate-level issue, I suggest you check out this answer which tells how to get started with RoboticsAcademy 
https://github.com/orgs/JdeRobot/discussions/233#discussioncomment-7983227

Feel free to play around with the application and file/solve any bugs you see

Tanishq30052002, 2024-03-04:
Hi @pawanw17 ,
I would like to work on the migration of a project, could you please assist with where to start if still there is work needed?
Thanks

closed 2024-07-01 by jmplaza