JdeRobot/RoboticsAcademy #2419: [New manager] Frontend support for new manager world reload mechanism

OscarMrZ, 2024-02-19:
Currently, the new startup sequence messages are only sent in the first world launch and not in the reload. This should be changed in [the world selector](https://github.com/JdeRobot/RoboticsAcademy/blob/master/react_frontend/src/components/visualizers/WorldSelector.js).

OscarMrZ, 2024-02-19:
Also, the RAM will have separate terminate functions for each level, so the frontend should be changed to send the appropiate new messages.

Surajpatra700, 2024-02-21:
Hey @OscarMrZ can you please tell me, what should i actually need to do right now.

OscarMrZ, 2024-02-23:
This was fixed at #2420. @Surajpatra700, we are currently undergoing a very intense manager refactorization. Please feel to test it (branch superthin-dev) and report any bugs or improvements you find.

closed 2024-02-23 by OscarMrZ