JdeRobot/RoboticsAcademy #2442: Review and Update the installation Instructions for Developers

AryanNanda17, 2024-03-08:
Since many users are facing difficulties in doing developer setup of RoboticsAcademy, there is a need to review and update the [Instructions for developers](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md).
@jmplaza , could you please suggest any changes and assign me on this issue. 
Thanks

OscarMrZ, 2024-03-29:
Hello @AryanNanda17. We are in the middle of intense refactoring, so the docs are changing quite a lot. If you'd like, please test the new docker compose installation and suggest any changes/issues you may find.

Apoorvgarg-creator, 2024-04-19:
@AryanNanda17, Using **Conda** to create Environment seems a nice update in the documentation.

javizqh, 2024-07-03:
Updating the naming scheme of RADI to RoboticsBackend has been done in this issue #2620

javizqh, 2024-07-12:
Maybe we could add how to create a GUI and a HAL using gui_interfaces and hal_interfaces

closed 2024-07-12 by jmplaza