JdeRobot/RoboticsAcademy #2449: AttributeError encountered when attempting to initialize beacons in the Position Control exercise

Arcane-01, 2024-03-22:
When attempting to initialize beacons using `HAL.init_beacons()` or retrieve their poses directly using `HAL.get_next_beacon().get_pose()` in the RADI version 3.4.30, an AttributeError is encountered. The error message indicates that the module 'HAL' does not have the attributed method 'init_beacons'.

[image]

ABD-01, 2024-03-25:
Are you still facing this issue??
Try restarting the container.

I went through the source code and there is `init_beacons` method available for the HAL in position_control_newmanager.
https://github.com/JdeRobot/RoboticsAcademy/blob/ee0191e083e0df8aaf52f11a9404fd2d18516ac1/exercises/static/exercises/position_control_newmanager/python_template/ros1_noetic/hal.py#L84-L93

Can you try printing `dir(HAL)` ??

Arcane-01, 2024-03-25:
Hi @ABD-01, even after restarting the container, I was still having the error. 
So, I made some changes to the `exercise.py` file in position_control_newmanger. 
Since the problem is resolved, I will be closing the issue.

closed 2024-03-25 by Arcane-01