JdeRobot/RoboticsAcademy #2461: Error on following the Installation for Developers(using Docker Compose)

Arcane-01, 2024-03-25:
Even after following the updated [Instructions for developers](https://github.com/Arcane-01/RoboticsAcademy/blob/Instr-for-Devs/docs/InstructionsForDevelopers.md) and trying to setup the developer environment using Docker Compose, I encountered the following error on trying to open the exercises. 

[image]

I was able to resolve this by creating a virtual environment, installing Django and its required packages, along with React dependencies, and finally running the development server for React frontend, as mentioned in the subsequent section `Local Installation`. Also there are some minor changes required in the current instructions. Should the instructions be updated accordingly?

OscarMrZ, 2024-03-26:
Hello @Arcane-01! Thanks for your suggestions. Currently, we are moving towards developing directly inside a container using docker-compose. The local installation will soon be deprecated, as a full django is shipper inside the container. 

Anyhow, in this case the problems probably were because the react frontend wasn't built. We forgot to add that to the docs, sorry for the incovenience! 

If you'd like to, you may check the new docs and dev procedure here: https://github.com/JdeRobot/RoboticsAcademy/pull/2463

pariaspe, 2024-03-29:
Fixed on #2463

closed 2024-03-29 by pariaspe