JdeRobot/RoboticsAcademy #2464: [Superthin templates] Fix GUI widget in follow line exercise

pawanw17, 2024-03-27:
The GUI Widget in the superthin version of the follow line exercise is broken, It is supposed to display a map overlayed on the camera images.

It works in master, but due to changes in superthin and movement from sharedVariables, the GUI just hangs up. The current workaround is to have a separate react component that only displays the camera images.

The files that likely need changes are 
https://github.com/JdeRobot/RoboticsAcademy/blob/master/react_frontend/src/components/visualizers/UpdateView.js
https://github.com/JdeRobot/RoboticsAcademy/blob/master/react_frontend/src/libs/UpdateRenderer.js

the development can be started at the 4.5.6.beta RADI
`docker pull jderobot/robotics-academy:4.5.6.beta`

This is how it is supposed to look
[image]

JovanYap, 2024-04-08:
Hi Pawan, I have downloaded the 4.5.6.beta development version and started it on my end and the overlay is not displayed as well. I am current taking a look at both the files and I will first create a separate react component that displays the camera image (I will try to find an alternative as well just in case this does not work) and conduct unit + integration testing after. Thank you!

JovanYap, 2024-04-09:
Hi @pawanw17  Thanks for the information! I see that the overlay is present but not updating on version 4.4.31 but it is not present for 4.5.6beta. I am currently taking a look at ws_gui.js code that is related to follow the line exercise since it updates the map overlay and the position of the car. Are you referring to the REST API code for the backend part and may I ask where can I view the code or changelog for the 4.5.6beta and 4.4.31 versions? Thank you!

pawanw17, 2024-04-09:
Hi @JovanYap |Unfortunately we don't have a good changelog, the backend code you can find here:
https://github.com/JdeRobot/RoboticsAcademy/blob/superthin-dev/exercises/static/exercises/follow_line_newmanager/python_template/ros2_humble/gui_exercise.py

4.5.6.beta 
involves migration of follow line exercise to superthin templates(removes exercise.py and refactores gui.py) 
It was supposed to use theses files
https://github.com/JdeRobot/RoboticsAcademy/blob/master/react_frontend/src/components/visualizers/UpdateView.js
https://github.com/JdeRobot/RoboticsAcademy/blob/master/react_frontend/src/libs/UpdateRenderer.js
but they were not working for this migration so I created a new temporary JS component just to render the images and nothing else
https://github.com/JdeRobot/RoboticsAcademy/blob/superthin-dev/exercises/static/exercises/follow_line_newmanager/react-components/SpecificFollowLine.js
https://github.com/JdeRobot/RoboticsAcademy/blob/superthin-dev/exercises/static/exercises/follow_line_newmanager/react-components/helpers/showImagesFollowLine.js

You can change the component the exercise uses from this file
https://github.com/JdeRobot/RoboticsAcademy/blob/390eed797fd8fcf3aa9532f81d8b41e779a9d1b5/exercises/templates/exercises/follow_line_newmanager/exercise.html#L23C38-L23C39

pawanw17, 2024-04-24:
Hi @JovanYap I took a look at the lap, map, v, w values and they seem to be static
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
{"image": "", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}

I don't think this would be the reason the widget crashes

Apoorvgarg-creator, 2024-06-12:
@pawanw17 The reason it was not rendering because the image is basically coming as empty "". 
The function used is `drawImage` getting nothing in image, hence no preview.

pawanw17, 2024-06-12:
@Apoorvgarg-creator even with the image the widget gets crashed....

Apoorvgarg-creator, 2024-06-12:
> @Apoorvgarg-creator even with the image the widget gets crashed....

Since, we are using Flex Container now, the size of the circuit is not fixed, which is scaling the image.

Yes, Since beginning _(Correct me if I am wrong)_  I remember while migrating also, image data was null only. we will basically draw a red dot on over the static image and use the position data coming from backend to move it on the map.

pawanw17, 2024-06-12:
@Apoorvgarg-creator If you run the code 
GUI.showImage(HAL.getImage())
Inside the while loop

Then you should see the image data being populated. 

I can share the samples tomorrow

pawanw17, 2024-06-13:
Hi @Apoorvgarg-creator 
The code I ran on 4.5.13
```python
from GUI import GUI
from HAL import HAL
# Enter sequential code!
HAL.setV(2)
while True:
    im = HAL.getImage()
    GUI.showImage(im)
    # Enter iterative code!
``` 

Here is the sample json
```
{"image": "{\"image\": \"/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAIBAQEBAQIBAQECAgICAgQDAgICAgUEBAMEBgUGBgYFBgYGBwkIBgcJBwYGCAsICQoKCgoKBggLDAsKDAkKCgr/2wBDAQICAgICAgUDAwUKBwYHCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgr/wAARCAHgAoADASIAAhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwDFaP8AyaY0XtVto/8AJpjRY7V/QB8aVWjz2zTDH/k1baLPbNMaP/JoAqtF7flTGjz2zVpovamtFntmgCo0f+TTWi9vyq00f+TTGi9qBlUx596aY8e1Wmiz2zTDH/k0AVWi9vyppj/yatNF7flTDHn3oAqtHjtimmMntmrTR49qYYvb8qBlUx/hTWj9RVox9v500xe35UAVTH6U0x+oq0Y6YYsdsUAVjH6GkMZ7irBjP1ppjA7GgCuYx9KaYz6A1ZMfvTTH6rQBWMY9CKQxnsasGP0NNMf+z+VAFcx+q00xj1qyYx7immM0AVzGfrSGP1WrBi/2fyppjHrQBXMY9aDGfWrBjPrTTH/s0AVzF/s0hj9jVgxj0IpPL96AK/l+9IYyfSrOw+tJ5f8AsigCuY/9kUnlf7P61Y8vH8NJ5f8AsmlZAQGP/ZpPK9jVgxgdc0mwetFgIPK9jR5Xsan8v3o8v3p2Ag8r2NHlexqfy/ejy/eiwEHlexpRH/s1NsHrS+WD0zSsBB5X+z+tKI/9kVN5f+yaXy8/w07AQCMj0pfL96n8v/ZFLsPrQBB5XsaURf7NTeX70ojHoTQBEIz60CMetTCP/ZpwjPrQBCI/RaURn6VKIx604R/7P50AQiMetOEfotSiM/SnCMe5oAi8s9zSiMehNSiL/Zpwj9TQBEIz6AU4Rj61KIv9n86cI/egCIRnsMUoj9TUoj9BmnCM/SgCIR+gpwj9alEWe2aeI6AIhH6CnCP8alEXt+dOEfv+VAEQjI7Ypwjz2zUoi9vzp4jz70ARCP8AyKcsXt+dSiPHtTxF7fnQK5EI8+9OEePapRH/AJFPEWO2KAIli9vzpyx/5FSrF7U8R/5FAEQjx2xT1i9vzqURY7Ypyxe1AiJY/wDIp4jx2xUoj/yKeIsdsUB0IVi9qeI/8ipVi9qeI/8AIoDqWGi/yaa0Xt+VW2i9qYYvb8qCSoYs9qY0Xb+dXGi/yaY0XtQO5UaL2/KmNFn3q20Xt+VNaL/JoC5TaL/JprRe35VbaL2phi9vyoAqNFn3prRVbaL/ACaYYsdqBlQxe35UxoqtmL2prRf5IoBFQxY9qYYvb8qtmL0FNMXsKAKhj/yaaYvb8qtmLFMMXt+VAyoYs00xexq20X+TTTF7UAVDFnsDTTFjsRVoxfSmmL2NAFUxZ7A00xY7EVaMWfSmmL2NAyqY/oaaYv8AZNWjFn0ppi9jQBWMfoaaYv8AZFWjH2/nTTF7UAVjH9RSGM+tWTHj1ppj+lAFYxf7NIY/Y1ZMX+zSGPHrQBW8v3pDGT1xVkx570hi9hQBWMX+zSGL/ZNWfK/2f1pPK9jQBX8sDrmk8v3qx5fvR5fvQBX8v3o8s9jU/lk9cUeV7CgCvsPqKNh9RVjyvYUeV7CgCvsPqKNh9RVjyvYUeV7CgCDyz3NHl+9T+V7CjyyOmKAIPL96Xyweman8v3o8v3oAgEX+yaURf7NT+V7Gl8r/AGf1oAg8sjpil8v3qcRewpRHjvQBAI/Y0oi/2anEefWlEX+zQBCIz60oj+pqYR49KcI8+tAEAi/2RThH6mphF7U4R/5FAEAi/wBk04R/QVMIvY04RY9KAIRFnsTThFjsBUwi9jThF9KAIRFnsTThFjsBUwi9jThFQBCIvaniLH/1qmEXt+dOEX+RQIhEXt+dOEf+RUwi9vzp4ioC5AIvb86eIs+9TCL2FOEXtQBCsdPEXt+dTLF/kCniL2oFchWKnCLHtUwiz2p6xf5FAEIi9vzpyxf5FTCL2/OnrF7UCIRFj2p4i9vzqZYv8inCL2/OgLkKxf5FPEWPaphFntT1i/yKAIVi9vzpyxf5FTLF7fnT1izQIstF7flTGiq40XtTWi9vzpXAptF7Uwxe35VcMXt+VMaL1pgU2i9qa0VXGi9qYYfQflQBTMXt+VMaL2q60Pt+dMaH2oApmGmGL2/Krhh9vyprQ+350AU2i9f1phiPpVww+1MMPtQMpmL2/KmtF/kirjQ+o/OmmH2oApmH0FMaL2q4YfammE+lAymYvamGL2q4YfQU1ofUfnQBTaL/ACaaYvb8quGH2prQ+v8AKgCmYv8AJFNMPtVww+1MaL1oAqGL1/UU0w+1XDF7U0w+36UDuVDF9aaYvYVbMPoP1ppi9f5UBcqGL2NIYv8AJFWjF7Uhix2NAFQxe1IYvY1bMPt+Ypph9v1oAqmL1/UUhiz2FWjFjsaQw+o/SgZV8r/Z/Wk8r2NWjF7CkMPt+tAFXyiOp/Sjy/erPkn3o8r2NAFXyvYUeV7CrPlH3/KjyfUfpQBW8r2FHlewqx5Q9vyo8oe35UAV/K9hR5XsKseUPb8qPKHt+VAFfyvYUeV7CrPk+g/Sjyj7/lQBX8v3o8onof0qz5XsaPJPvQBX8r2NL5X+z+tWPJ9v1pREPagCv5WOwpRF6foKsCH0H6UohPvQIriL2NKIvarHk+3604Q+36UDK4i9P0FKIvY1YEWexpRF7UCIBF7CnCI+9TiH0H6U4Q+o/WgCuIvb86cIvT9BVgReg/IU4Re1AXK4h9qcsX+RVhYfT+VOEPtQIriL2/OnLF/kVYWL2/IU4Q+350AVxF7U8RVOsPoPypwh9RQBAIaeIvb86nEPtThD7UAQLF/kCnCL2/OrAh9RTlh9B+VArkAi9qesXp+lTrD7U4Q+1AXIBF7fnT1i9qnWL0/SnCL2/OgCBYqeIvb86nEJ9KesPt+VAiBYvanLF7flU4h9RT1i9qAIBF7fnT1i9qnWL2/KnCL2/OgCARU8Re351OIqcsXtSuBaaH2phh9vyq60NMaH2/MVNwKbQ+ophhPpV0w+gpjQ+1FwKRh9BTWh9R+dXWh9RTDD6U0wKRh9qa0PtVxofUU0w+1O4FJofammH2/Krhh9Ka0PtRcCkYfammH2q6YfamND7UwKRh9KY0PqP0q8YfamGH0oApGH2pph9qutD7fpTTD7UAUmh9R+lMMPpV0w+1NMPtQBSMPt+lNMPpVww0hh56fpQMpGH2/SmmH0FXDD7U0w+36UAUjD7Uhh9v1q4YfSmmHnp+lAFMw98fpTTD6VcMNIYfagZTMPt+lNMIz0FXDD7UhhoBMpmH2/WmmHnp+lXDBz0/SkMIz0oApmEZpDD6D9auGGmmDnp+lAXKhhPvSeT7fpVwwe36U3yR7UAioYe2BSeSParhhpPJPvQFyp5I9P1o8n2/WrZhJ7Unk+36UAmVPJPvR5J96t+T7fpR5Pt+lAyp5J96PJPvVvyfb9KPJ9v0oAq+T7frR5I9P1q15Pt+lKIcdqBXKnkj2pRD2wKteSfelEPNAyr5Pt+lKIT71Z8ke1OEHt+lAiqIfUfrSiEZqyIOen6U4Q0BcqiHnp+lOEPt+tWRDz0pRBz0/SgCsIRnoKcIfb9KsiHmlEPtQBWEPrThD3x+lWRDx0pRDQK5XEPt+tKIfb9KsiHnp+lOEPrQBXEPtThD7fpVgQe36U4Q+1AFcQ+tOWH2/SrIh9v0pRD7UAVxD6j9aeIfb9KsCH2pwh9qAK4h9qcIfarIh9qcsPt+lAisIfQfpTxD7fnVgQ+tPEPtQBXEPtThD7VYEPtTxD6igCuIfb86csPt+VWFh9qcIfUUAQCH2pyw+1WFh9qcsPoKVwIFh9B+VOEXt+dWBD608Q+1K4FdYfanrD7flU4h9qeIfUUrgVxD7fnT1h9qnWH2p4houBaaH2phh9qvND7fpTGh9qVwKTQ+1NMPtV1ofamtB7UAUTD7U0w+36VdMNNMHtQBRMPpTDD7VeaH2prQ0AUTD7U1ofarpg9v0pphp3AotD7fpTTD6Crpg9qaYPai4FEw+1NMPtV5oPamGD2/SncCkYaaYfb9KumH2ppg9v0p3AomH2ppg9v0q8YaYYPai4FIw00we36VdMFNMHtTApGGmGD2q8YfammDvigCiYM9qQwe36VdMFIYPagCiYaaYPb9KvGD/9VNMHt+lAFEw+1IYM9qumCkMHtQBSMHt+lNMNXjB7U0wUDKRgz2pPI9v0q6YPakMHt+lAFLyMdRSGGrvkH+7SeR70AUjBnqKPs/tVzyB/do+z+1AXKXk+36UGDPb9KuGDHb9KPIHp+lAXKX2ceh/Kj7OPQ/lVz7OPQ/lR9nHofyoC5T+zj0P5UfZx6H8qufZx6H8qPs49D+VAXKnkY7fpR5Pt+lXPIHp+lHkZ7fpQFyp9n9qPIx0FXPs/tR5A9KAuVBDSiDParfke9L5B/u0AVPI9v0pRBjtVsQe36Uog9qAuVBDThB7fpVoQU4Qe1AFQQY7Uoh9qtiD2pRBQIqiD2/SnCGrQg9v0pwgoAqiD2/SlEGO1WxB7Uogx2oAqiD2p4hqyIO+KcIfagCsIPb9KcIasiDvinCCgCssHtTh [... cut ...] M8pby5mLD3qRVbbhDgUDHmywxMecA8UDfEORUBlniP3zj0zUhjknGVlP50AOa9xw3TvULWsUzb4mwakFjMOWGfrTLhjEmY1CsP1oESpbzKuDyKimgeNvMj4I6j1qSx1SWVdjwjI9qLia4Rt3k5H0oAWGVLuPa33hUMkb275A4p4kWY7xDtbvilZmYYY5+tAyHyjnzIT9V9KXzGAxItPCheQMUp560CIDAkx4OKt20s0KeTP8AOh6E1FsUHIWkLSLwGOKAFkH2aXzIiQpPNWQnnRebEN2OoqqhLfKTn2NPUmIYQlT7UDJFUSqTE3I6g9RUbu8bYYVBOLgN5qSHPrRa6sxb7NewhwejY5oKTuWFYSD5WwaVXw21ximvalG82Enaexp7eekXmJGHx2xQM//Z\", \"shape\": [480, 640, 3]}", "lap": "0", "map": "(151.0, 77.0)", "v": "0.0", "w": "0.0"}
```

Apoorvgarg-creator, 2024-06-13:
@pawanw17 Please confirm if this json is sent when **update** state.

pawanw17, 2024-06-13:
@Apoorvgarg-creator yes

pawanw17, 2024-06-15:
@Apoorvgarg-creator I tried to go back to the old widget by updating this line https://github.com/JdeRobot/RoboticsAcademy/blob/23ebd74ae417596e97cefb3fe37147ca363ffc89/exercises/templates/exercises/follow_line_newmanager/exercise.html#L23
with
{% react_component components/visualizers/UpdateView %}{% end_react_component %}

And it seems the issue is intermittent now, the widget works most of the time but hangs abruptly sometimes or doesn't appear.

Apoorvgarg-creator, 2024-07-01:
@pawanw17 I ran the code sent by you. I put a debugger also, but I didn't get any data.

Apoorvgarg-creator, 2024-07-02:
@pawanw17  Check this, No data is coming from backend in image.

[image]

[image]

pawanw17, 2024-07-02:
The data is not going because your code is stuck probably because of a runtime error see the console in UI, there is some error in your online editor code

closed 2024-09-09 by CDAM2020