JdeRobot/RoboticsAcademy #2465: Dependency tree issue [dependencies version in package.json deprecated]

vaishnavimulik25, 2024-03-28:
[image]
I have followed the latest documentation and used the latest script for installation. Since I was getting this issue I tried manually running commands in develop.sh script. I found error during npm install.

vaishnavimulik25, 2024-03-28:
[Modified develop_academy.sh](https://github.com/JdeRobot/RoboticsAcademy/pull/2466)

OscarMrZ, 2024-03-29:
Hi @vaishnavimulik25, thanks for your time! Please describe your issue in more detail, why are you using npm? RA uses yarn as the package manager, and should always be used (as it is in the script). Do you also have errors with yarn?

Regarding

vaishnavimulik25, 2024-03-30:
Yes I had errors with yarn previously but after using [updated develop_academy.sh](https://github.com/JdeRobot/RoboticsAcademy/commit/cfa60b532e5fbb7e6d4a0245dd711925ffc91b79) I didn't had error.

OscarMrZ, 2024-04-22:
Outdated due to big changes in main branch. Please feel free to reopen if you think we are missing something!

closed 2024-04-22 by OscarMrZ