JdeRobot/RoboticsAcademy #2471: Error in adding new exercise section

vaishnavimulik25, 2024-03-30:
I was following the Adding new exercise section in  [InstructionsForDevelopers](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md) While running "python3 manage.py runserver 0.0.0.0:7164", I got this error. 
[image]

I checked the [requirements.txt](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/requirements.txt) and pylint version was not specified. I have worked on this issue. I figured out pylint==2.17.4 version works for it.

pawanw17, 2024-04-08:
Hi @vaishnavimulik25 Can you make a PR for it?

vaishnavimulik25, 2024-04-09:
Hi @pawanw17 Yes I will make PR immediately.

pawanw17, 2024-04-11:
fixed by #2479

closed 2024-04-11 by pawanw17