JdeRobot/RoboticsAcademy #2484: [Rescue people] Restart button doesn't work

agonzc34, 2024-04-21:
In the exercice Rescue people when I click on the restart button, it resets the world models, but it outputs the next log in the docker terminal and it does not allow me to run my code.
Working on 4.5.7.BETA.

12:48:33 [MainThread  ] [ERROR] (my_app_logger)  [Errno 32] Broken pipe
Traceback (most recent call last):
  File "//RoboticsAcademy/src/manager/manager/manager.py", line 466, in start
    self.process_messsage(message)
  File "//RoboticsAcademy/src/manager/manager/manager.py", line 364, in process_messsage
    self.gui_server.send(message.data)
  File "/RoboticsAcademy/src/manager/libs/applications/compatibility/server.py", line 39, in send
    self.server.send_message(self.current_client, data)
  File "/usr/local/lib/python3.10/dist-packages/websocket_server/websocket_server.py", line 78, in send_message
    self._unicast(client, msg)
  File "/usr/local/lib/python3.10/dist-packages/websocket_server/websocket_server.py", line 195, in _unicast
    receiver_client['handler'].send_message(msg)
  File "/usr/local/lib/python3.10/dist-packages/websocket_server/websocket_server.py", line 344, in send_message
    self.send_text(message)
  File "/usr/local/lib/python3.10/dist-packages/websocket_server/websocket_server.py", line 413, in send_text
    self.request.send(header + payload)
BrokenPipeError: [Errno 32] Broken pipe

OscarMrZ, 2024-04-22:
Does this still happen with the last RADI?

pawanw17, 2024-04-27:
Can confirm that the error exists in latest humble-devel,but even after the error I am able to run the code as expected

agonzc34, 2024-04-28:
I have tested on 3.4.30 and on 4.5.9 docker images.
On 3.4.30 the error log doesn't show and it allows me to restart the exercise, but when I restart the simulation, the drone starts flying randomly.
On 4.5.9 it doesn't even start, it logs this:
[image]

pawanw17, 2024-04-28:
Hi @agonzc34 the issue in 4.5.9 is fixed in master and you will be able to run the exercise without issue in the upcoming 4.5.10

regarding 3.4.30, we are aware of that issue but ROS1 RADI is no more in active development

jmplaza, 2024-04-30:
Hi @agonzc34,
 
RADI 4.5.10 has been released. Drone-rescue-people exercise is working fine again. Could you please check it?

javizqh, 2024-07-03:
With 4.6.1 there is no error when resetting but the drone starts flying even if the code is empty after resetting the exercise.

javizqh, 2024-07-03:
@pariaspe Do you know what could be making the drone not reset itself properly when clicking the reset button, but working properly when changing to another exercise and the starting this exercise again?

pariaspe, 2024-07-04:
> @pariaspe Do you know what could be making the drone not reset itself properly when clicking the reset button, but working properly when changing to another exercise and the starting this exercise again?

My guess is that PX4 (the autopilot simulated within gazebo) is not well reset

pawanw17, 2024-07-04:
@javizqh @pariaspe I am certain that this didn't happen before.

pawanw17, 2024-07-04:
I am closing this issue and opening another one.

closed 2024-07-04 by pawanw17