JdeRobot/RoboticsAcademy #2494: Update exercises docs with docker compose launching

Syekeon, 2024-04-25:
It could be a good idea adding the docker compose launching in those exercises already working with the last versions.

Syekeon, 2024-04-25:
Also I could be a good idea changing the docker launch without the ports that aren't using in the last versions (or adding a second launch in docs depending on the RADI version)

jmplaza, 2024-04-27:
AFAIK only the mapping of 4 ports is required now. For instance:
`docker run --rm -it -p 7164:7164 -p 6080:6080 -p 1108:1108 -p 7163:7163 jderobot/robotics-academy:4.5.9`

- 7163 for connection between the exercise web page and the RAM
- 6080 vnc websocket for Gazebo viewer
- 1108 vnc websocket for console viewer
- 7164 for local webserver

pawanw17, 2024-04-28:
Hi @Syekeon In amazon warehouse exercise I have turned reverted to lift() and putdown()... so let's keep it that way... you may have to update the PR _pages/exercises/MobileRobots/amazon_warehouse.md

Syekeon, 2024-04-28:
@pawanw17 Done it!

PR #2504 (2024-04-29): Exercise docs update
fixes #2494
closed 2024-04-29 by Syekeon