JdeRobot/RoboticsAcademy #2506: Move interfaces to Robotics Infrastructure repo (hal_interfaces)

pawanw17, 2024-04-28:
- [ ] Follow line
- [ ] Global Navigation
- [ ] Obstacle Avoidance
- [ ] Basic vacuum cleaner
- [ ] Autoparking
- [ ] Follow person
- [ ] Localized vacuum cleaner
- [ ] Rescue people
- [ ] Amazon Warehouse
- [ ] Montecarlo laser loc 
- [ ] 3D reconstruction

jmplaza, 2024-04-28:
`hal_interfaces` is something very related to RoboticsAcademy and not to other general robotics applications. It is part of the templates of the RA exercises. What about putting `hal_interfaces` in a directory named `common` (or something similar) in the RoboticsAcademy repo itself?

pawanw17, 2024-04-29:
Yes, that sounds better... we can also move console.py and a genericGUI.py their and have the GUI.py in the exercises be a child class of genericGUI.py

Some of the hal_interfaces have already moved to RI though, https://github.com/JdeRobot/RoboticsInfrastructure/tree/humble-devel/hal_interfaces/hal_interfaces

pawanw17, 2024-05-01:
Hi @jmplaza, On digging further I found that @OscarMrZ has made ros nodes out of the hal interfaces which gets compiled with other ros2 nodes, custom robots etc, I think that is a good way of doing things... what do you think?

javizqh, 2024-06-05:
- [X ]  Global Navigation
- [X ]  Obstacle Avoidance
- [X ]  Basic vacuum cleaner
- [ ]   Follow line
- [ ]   Autoparking
- [X ]   Follow person
- [X ]   Localized vacuum cleaner
- [X ]   Rescue people
- [ ]   Amazon Warehouse
- [ ]   Montecarlo laser loc
- [ ]   3D reconstruction

closed 2024-06-12 by pawanw17