JdeRobot/RoboticsAcademy #2535: [Global Navigation] Runtime error when running example code

pawanw17, 2024-05-22:
When trying to run the solution code: https://github.com/JdeRobot/RoboticsAcademy-solutions/blob/master/solutions/global_navigation/basic_Global_Navigation.py
```
root@e14a8c465c4c:/# HAL-Nodes Thread Started
World : [-108.91089108910893, 1.6248839368616643]
Error in shared image add TypeError("'NoneType' object does not support item assignment")
Traceback (most recent call last):
  File "/workspace/code/shared/image.py", line 180, in add
    self.shm_buf_1[:] = image.tobytes()
TypeError: 'NoneType' object does not support item assignment
``` 

[image]

javizqh, 2024-06-03:
Are you able to reproduce this bug? I am not able to do so

javizqh, 2024-07-03:
Can you reproduce this in the new 4.6.1 RB? @pawanw17

dduro2020, 2024-07-10:
I've been trying this solution and I can't reproduce the issue
. But with the solution https://github.com/JdeRobot/RoboticsAcademy-solutions/blob/master/solutions/global_navigation/Global_Navigation.py I get a similar error message:
```bash
Error in shared image add IndexError('mmap slice assigment is wrong size')
Traceback (most recent call last):
  File "/workspace/code/shared/image.py", line 156, in add
    self.shm_buf_2[:] = image.tobytes()
IndexError: mmap slice assigment is wrong size
```

[image]

javizqh, 2024-08-02:
This can be closed as this issue cannot be longer reproduced.

closed 2024-08-02 by javizqh