JdeRobot/RoboticsAcademy #2544: [Obstacle avoidance] Not receiving images from camera

javizqh, 2024-06-03:
It does not receive the images from the camera topic and it gets stuck infinitely

javizqh, 2024-06-03:
It happens randomly

javizqh, 2024-06-03:
If someone can test it a couple of times it would be helpful.
I only manage to experience this behavior once, the first time I launched the docker. The rest of the times it would work perfectly

jmplaza, 2024-06-03:
Maybe in this exercise we could even remove the camera, as it is not required. The solution for this challenge only requires laser and position to work. The onboard vision definitely is nice and cool, but it takes a significant computational cost... and so reduces the Gazebo RTF. 

What do you think?  @pawanw17 @javizqh

javizqh, 2024-06-03:
Here is a video of what the change could be

https://github.com/JdeRobot/RoboticsAcademy/assets/124056014/a01fc85c-5c1f-4275-b12b-43ac0410ab33

https://github.com/JdeRobot/RoboticsAcademy/assets/124056014/9af4ad56-a188-48a0-8ea6-76ce2029870c

javizqh, 2024-06-03:
Or if we want the proportions to remain to not widen it, it could look like this:
[image]

jmplaza, 2024-06-03:
Cool. I would set the background color in the upper image to gray, more similar to the pavement. That way the only blue things are the laser readings, both in the upper image and in the Gazebo Viewer

javizqh, 2024-06-03:
So something like this would be the new UI:

https://github.com/JdeRobot/RoboticsAcademy/assets/124056014/e7d7393d-31a4-4d55-9e97-6b89f8073fd5

javizqh, 2024-06-03:
[image]

pawanw17, 2024-06-04:
@jmplaza Does this look alright or we should spread it as @javizqh had suggested?
[image]
am not a big fan of yellow

jmplaza, 2024-06-04:
:-) let's spread the widget, and let the orange be only in the top horizontal bar

javizqh, 2024-06-04:
Which one of the 3 is better:
[image]
[image]
[image]

jmplaza, 2024-06-04:
The first one, from my point of view

jmplaza, 2024-06-04:
Instead of 3.5m, maybe we could draw it at 5m, by the way.

javizqh, 2024-06-04:
The max laser distance is 6.8 meters, so maybe with a 5m box it goes outside of the border of the widget

pawanw17, 2024-06-04:
@javizqh Can you please check and update? if 5m box goes out?

closed 2024-06-04 by pawanw17

javizqh, 2024-06-04:
It fits perfectly

[image]

PR #2545 (2024-06-04): [Obstacle avoidance] Updated excercise 
This resolves #2544.
Also has the following changes:

- Removing interfaces in order to use hal_interfaces
- Removing odometry from map
- Removing image because of what is explained in #2544 
- Using real distances for the areas displayed
- Removing unnecessary threads in the GUI