JdeRobot/RoboticsAcademy #2551: [Localized Vacuum Cleaner] On reset last position is mark as cleaned

javizqh, 2024-06-05:
When reseting the excercise the last point will remain.

Here is a trace of console logs to see what is happening:

- At the top is the last position of the previous iteration
- Reset
- On the first iteration the Specific js from the excercise tells the helper to draw in one position but in the helper the values received are from the last execution.

[image]

Apoorvgarg-creator, 2024-06-23:
@javizqh Just to get clarity over the issue, The problem is that on resetting it is still giving you the last iteration position.
Where as it should be initial position

javizqh, 2024-06-23:
If it was the initial position it would be fine, preferably it shouldn't give a position from the last execution.

Apoorvgarg-creator, 2024-06-23:
> If it was the initial position it would be fine, preferably it shouldn't give a position from the last execution.

Thank you for the reply. working on the issue, So Sorry for the late view.

Apoorvgarg-creator, 2024-06-23:
[image]

@javizqh From the above Screenshot, I wanted to ask in **vacuum_cleaner_loc_newmanager** 
The **SpecificVacuumCleaner** widget you uses is from **exercise/vacuum_cleaner_newmanager/SpecificVacuumCleaner**

Apoorvgarg-creator, 2024-06-23:
@javizqh Can you share me the solution you ran.

Apoorvgarg-creator, 2024-06-23:
> <img alt="Screenshot 2024-06-23 at 11 03 23 PM" width="1430" src="https://private-user-images.githubusercontent.com/57873504/342102848-3033a92d-9960-4f8d-8ba6-f6a1b6bc8796.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxNjQzNjksIm5iZiI6MTcxOTE2NDA2OSwicGF0aCI6Ii81Nzg3MzUwNC8zNDIxMDI4NDgtMzAzM2E5MmQtOTk2MC00ZjhkLThiYTYtZjZhMWI2YmM4Nzk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIzVDE3MzQyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyYjY3MjVjNGM1YTQzZDc2MDVlNzAwOTkxZTBiMGM4NDliMDc3NTk0OGQ4YzIyZTNkZjkxMGVhMDg3OWJkYzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tLKaTkY-D3Pe8IKydPP7RiZQr1q79PEHMOcJoG2InBg">
> @javizqh From the above Screenshot, I wanted to ask in **vacuum_cleaner_loc_newmanager** The **SpecificVacuumCleaner** widget you uses is from **exercise/vacuum_cleaner_newmanager/SpecificVacuumCleaner**

I think the file needs to point at 
1. _**exercise/vacuum_cleaner__loc_newmanager/SpecificVacuumCleaner**_
2. _**exercise/vacuum_cleaner__loc_newmanager/BasicVacuumCleaner**_

jmplaza, 2024-06-23:
> @javizqh Can you share me the solution you ran.

I guess you can find it in the [RoboticsAcademy-solutions repo](https://github.com/JdeRobot/RoboticsAcademy-solutions) (private)

Apoorvgarg-creator, 2024-06-23:
> > @javizqh Can you share me the solution you ran.
> 
> I guess you can find it in the [RoboticsAcademy-solutions repo](https://github.com/JdeRobot/RoboticsAcademy-solutions) (private)

I ran the code from solution.py present in RoboticsAcademy-solutions repo for this particular exercise but got some error in the console.

javizqh, 2024-06-23:
Sorry, there is a PR open in the solutions repo with the new solution.

jmplaza, 2024-06-23:
Merged @javizqh

PR #2595 (2024-07-02): Fix: Issue 2551
Fix #2551