JdeRobot/RoboticsAcademy #2583: [3D reconstruction] Does not work on Chrome or Chromium based browsers

javizqh, 2024-06-20:
The 3D canvas that appears on Firefox does not appear on Chrome or Chromium based browsers causing the exercise to crash when you execute GUI.ShowNewPoints() a couple of times.

Firefox
[image]

Chromium or Chrome
[image]

javizqh, 2024-06-20:
In Chromium the width of the canvas is set to 0, while in Firefox it has an actual width.

Same happens with the images from the exercise. In Firefox they fit within the UI, but in Chromium the images appear bigger than they should, not displaying the image properly

javizqh, 2024-06-20:
After the PR the issue that remains is that the camera images are not displayed properly on chrome:

# Chrome:

[image]
[image]

# Firefox:

[image]

Apoorvgarg-creator, 2024-06-23:
@javizqh Are you working on this issue ? I see you have raised one PR and still some problem is occurring on this.

If possible, we can work together on this issue.

javizqh, 2024-06-23:
I made a quick fix in order to have the exercise working at least.

pawanw17, 2024-06-23:
Unfortunately, I never faced this issue, I used Chrome on Ubuntu 20.04 while testing

Apoorvgarg-creator, 2024-07-02:
Hi @javizqh For me It is running perfectly fine in Chrome

javizqh, 2024-07-02:
Tell @pawanw17 to test it, if It works for him we can close this issue

pawanw17, 2024-07-02:
Looks fine to me:
[image]

closed 2024-07-02 by pawanw17