JdeRobot/RoboticsAcademy #2594: Update Robotics Academy Image in compose_cfg

Apoorvgarg-creator, 2024-06-23:
Currently, The latest development version of RAM is **4.5.13**

But In **Yaml** files present inside **compose_cfg** is pointing to **4.5.9**

jmplaza, 2024-06-23:
??According to the [RAM repo](https://github.com/JdeRobot/RoboticsApplicationManager/releases), 5.2.0 is the last packaged release of RAM. 

Those numbers you mention seem RADI release numbers...

Apoorvgarg-creator, 2024-06-23:
> ??According to the [RAM repo](https://github.com/JdeRobot/RoboticsApplicationManager/releases), 5.2.0 is the last packaged release of RAM.
> 
> Those numbers you mention seem RADI release numbers...

Yes, Accidently I typed RAM. Apologies for the mistake.
I meant RADI

PR #2596 (2024-07-02): Fix: Update RADI latest version in compose_cfg
Fix #2594
closed 2024-07-02 by pawanw17