JdeRobot/RoboticsAcademy #2606: [Amazon Warehouse] Add getSimTime() function in HAL module

lu164, 2024-07-01:
HAL.getSimTime() function returns the simulation time

pawanw17, 2024-07-03:
Hi @lu164 why is this method needed?

jmplaza, 2024-07-03:
For an advanced way of Path Planning in [OMPL](https://ompl.kavrakilab.org/), a way that takes into account control restrictions of the mobile robot. With it OMPL delivers a temporal sequence of lineal speed and rotation speeds (V,W) as the output plan. 

That plan execution requires indexing in the sequence using time, "real" time. In fast machines, with Gazebo RTF close to 1, the wall clock time and the simulated time are quite similar, but in slow machines they diverge soon. In such slow machines, using the wall clock time results in a wrong plan execution, but using simulated time provides better plan executions and paths.

PR #2618 (2024-07-03): [Amazon Warehouse] Adding SimTimeNode and getSimTime
Adding SimTimeNode and getSimTime. Closes #2606
closed 2024-07-03 by pawanw17