JdeRobot/RoboticsAcademy #2665: The “Learn More” button does not work correctly in certain Robotics Academy exercises’ cards

mariabusto, 2024-07-22:
The “Learn More” button does not work correctly in certain Robotics Academy exercises’ cards when selected to see the tags of those exercises. Specifically, when this button is selected, the following errors show in the developer’s console and the webpage becomes white. 

[image]

This error happens in the “Learn Button” of the cards of the following exercises: Vacuum Cleaner; Follow Person; Global Navigation; Amazon Warehouse; and Montecarlo Laser Loc.

Apoorvgarg-creator, 2024-07-29:
Thank you @mariabusto.

Apoorvgarg-creator, 2024-07-29:
The Root cause of the issue is 

The correct format for tags are - {"tags": ["ROS1", "ROS2","AUTONOMOUS DRIVING"]} ✅

the square brackets are currently missing in the mentioned exercises. Hence it was throwing error.

I look at Global Navigation data here for tags as {"tags": "ROS2"}. ❌

Apoorvgarg-creator, 2024-07-29:
Putting the solution here - 

Since there was no option in Django for Array datatype we had to use this approach. But Instead of hardcoding it on format, we will fix it in the react_frontend code where we will check if it can be converted into array or not.
If can be taken as an array, then we will simply map. Otherwise, we will first make an array using **"comma"** as the separator.

closed 2024-07-29 by Apoorvgarg-creator