JdeRobot/RoboticsAcademy #2668: Vnc views freeze when launching script for developers with nvidia acceleration

dduro2020, 2024-07-24:
No viewers are updated and it is not possible to interact with them, I want to stress that when launched with user instructions there is no problem

[image]

dduro2020, 2024-09-02:
The vnc views now load correctly, but whenever you change an exercise world, apart from adding up errors from trying to finish processes that have already finished, you get an error when supposedly nothing has failed.

Here's a screenshot of the error message (just after it loads correctly, but it looks like a residual message that shouldn't appear)
[image]

First time changing the world:
[image]

Second time (And each time you do it again, error messages are added up, which you will see again when the container is finished):
[image]

javizqh, 2024-09-02:
The error displayed to the user does not matter because it is caused by the Gazebo viewer loading a couple of seconds slower than the console and it is solved after a 2 or 3 seconds

javizqh, 2024-09-02:
I'm going to close this issue and open it in the Manager repository

closed 2024-09-02 by javizqh