JdeRobot/RoboticsAcademy #2753: Need help! noVNC error

jmplaza, 2024-09-30:
### Discussed in https://github.com/JdeRobot/RoboticsAcademy/discussions/2725

<div type='discussions-op-text'>

<sup>Originally posted by **codezerro** September 19, 2024</sup>
I got this error when run jderobot/robotics-backend docker image.
but running jderobot/robotics-academy image without any error.

My Pull Request on [RoboticsAcademy-Desktop](https://github.com/JdeRobot/RoboticsAcademy-Desktop) is [here](https://github.com/JdeRobot/RoboticsAcademy-Desktop/pull/16)

[image]
</div>

codezerro, 2024-10-01:
Hi @jmplaza , I got another error; here it is. 

[image]

I was working hard to figure out the solution. End of the day, I found this error occurs because of browser network cache. I disable cache, and it works without any issue. This is the only solution I've found, and it works.

I also added some code to the [RoboticsAcademy-Desktop](https://github.com/JdeRobot/RoboticsAcademy-Desktop/pull/16) repo, which helps disable the HTTP cache.

Anithpavan, 2025-04-14:
Hi! I'm Anithpavan, a GSoC 2025 applicant. I'm interested in looking into this issue as part of my first contribution. I'll try to reproduce the error and investigate what’s causing the noVNC failure in the backend image. Any initial guidance or references would be greatly appreciated!

codezerro, 2025-04-14:
Hi @Anithpavan, 
Thanks for your interest.

We were developing the RoboticsAcademy desktop, which is based on Electron JS. This error happened because of Electron JS caching. We solved this issue. 

I'm going to close this issue. 

You can contribute to other issues.
Glad to hear that you are participating in GSOC this year. Good luck. 👍🏽

closed 2025-04-14 by codezerro