JdeRobot/RoboticsAcademy #2816: [Global Navigation] Question about map image having gray borders

franmore-urjc, 2024-10-24:
In the Global Navigation exercise, the image we get from the map represents the free space as white (value 255) and the obstacles/buildings as black (value 0).

However, the borders of the mage (the first/last row/column) are not white or black, but have some random gray values between 10-40.
Is there a reason for these pixels? I am not sure if this is a bug or if it was intentional, and if some code assumes the image is binary (free / obstacle), it may run into trouble.

Thanks in advance.

franmore-urjc, 2024-10-24:
More information: It seems the gray pixels are not only at the image borders, but everywhere:

[image]

These might have been created when resizing the image.

javizqh, 2024-10-24:
Probably it is because of the resizing. I will try to paint them on top

javizqh, 2024-10-24:
Can you assure me that the new one is what you wanted?

franmore-urjc, 2024-10-24:
Awesome! Looks much better now, thank you.

closed 2024-10-24 by javizqh