JdeRobot/RoboticsAcademy #2817: [Global Navigation] Clicked target is deleted after starting exercise

franmore-urjc, 2024-10-24:
When clicking in the "play" button to run the exercise, the target that was marked in the GUI is removed (cleared). In previous versions the target stayed in the GUI after clicking "play".

It is ok to clear everything after clicking the "reset" button, but removing the target after clicking "play" is a bit painful for the user, because it requires setting a target while the exercise is running. Before, one could set the target and then run the exercise, which was much easier IMHO.

javizqh, 2024-10-24:
Can you check in the PR if that is the behaviour that you expected. If so, then I will merge it

javizqh, 2024-10-24:
Did you launch the docker using the developer script?

franmore-urjc, 2024-10-24:
Yes, I did. I can also confirm that issue #2818 is solved by PR #2820.

If you need more tests from my side, I won't be able to do them until tomorrow, but let me know if you need anything else to trace it.

javizqh, 2024-10-24:
If you could share I video of the bug I can try to check what is going wrong, because on my end everything seems fine

franmore-urjc, 2024-10-25:
Video of the issue:
https://github.com/user-attachments/assets/c4511305-993c-4609-a058-26e517f060a9

As you can see, when I click the play button to start the exercise, the target is removed from the GUI. It is only possible to select a target once the exercise hasstarted.

javizqh, 2024-10-25:
Can you reupload the video in a valid format?

javizqh, 2024-10-25:
> Sorry, Github created it that way when attaching the file. I see that the online visualization does not work, but if you right-click the link and do a "Save link as...", then you can download the video and it works.
> 
> If this doesn't work for you let me know and I will upload it somewhere else.

Thaks. I will try to solve it this afternoon

javizqh, 2024-10-25:
This is what i looks to me with the latest changes

https://github.com/user-attachments/assets/083e6426-6d9b-4988-af89-13edef083f7f

Can you check if it works for you?

franmore-urjc, 2024-10-25:
The latest commit in the branch for this PR was 19h ago. I recorded the video using that commit.
If you changed anything else today, that's not pushed yet upstream.

javizqh, 2024-10-25:
I also used that commit to record the video. Can you check in other browser or refreshing the cache?

dduro2020, 2024-10-25:
After testing it, it works perfectly, now the targets are not reset when pressing play and no traces are printed that have not been written by the user.

franmore-urjc, 2024-10-25:
I just tried it in google chrome. I didn't have chrome before, so it was a fresh install and the cache was empty. I still have the same issue...

I also checked the code inside the container to verify that I am using the correct patch and that the files are being correctly mounted into the container. The files are identical, so I'm sure that I'm using the last version.

I'm not sure if I am missing anything else. We will start this exercise on Monday, so if these changes are merged and published, then I will let you know if anyone else has the same issue. If I am the only one, I don't mind.

Thanks for the support.

javizqh, 2024-10-25:
Ok, then I will merge it. If the problem still remains hen we release the next version of Unibotics, please reopen this issue.

Just to confirm, to check the frontend you are not using Unibotics @franmore-urjc

franmore-urjc, 2024-10-25:
Wellllll, I did't know that affected and I was, in fact, using Unibotics! It makes sense now, thank you for the check!

I just tried again and it works as expected. The only thing to take into account is that in the beginning of the exercise, the method `GUI.getTargetPose()` returns `None`, even if there is a target selected. This is a bit weird, but I assume this is beause of the communication between react and the python code.

Anyway, thank you again for your help.

javizqh, 2024-10-25:
I will close this and open a new issue for that https://github.com/JdeRobot/RoboticsAcademy/issues/2823

closed 2024-10-25 by javizqh

javizqh, 2024-10-25:
I will try to solve that. Thanks

> Wellllll, I did't know that affected and I was, in fact, using Unibotics! It makes sense now, thank you for the check!
>
> I just tried again and it works as expected. The only thing to take into [...]

Sadly it depends on react, but normally it will work after 1 or 2 seconds