JdeRobot/RoboticsAcademy #2909: Harmonic processes do not always die

dduro2020, 2024-12-11:
The harmonic and bridge launchers for some reason don't die in the usual way, so we have to kill them by force. When you exit the harmonic exercise after playing it and without resetting, the processes do not die. This causes a crash and the docker has to be relaunched. 

This issue has to be resolved from RAM.

isatyamks, 2024-12-12:
Could you provide more details on why the harmonic and bridge launchers aren't terminating properly after exiting the harmonic exercise? Specifically, what causes the processes to remain in memory and how they can be forcefully killed?

can we use this to stop that harmonic exercise

def cleanup_processes(signum, frame):
    os.system('killall -9 harmonic_process_name')
    sys.exit(0)

signal.signal(signal.SIGTERM, cleanup_processes)

dduro2020, 2024-12-12:
We don't know why they don't finish, but we think it may be an issue with the gazebo harmonic itself. So we decided to kill them by force. Right now we already have a function that kills the harmonic processes and even their childs so that there are no zombies left, the problem is that in the situation I mentioned, they are not eliminated, I don't know why.

javizqh, 2024-12-12:
Duplicate https://github.com/JdeRobot/RoboticsApplicationManager/issues/178

closed 2024-12-12 by javizqh