JdeRobot/RoboticsAcademy #2916: Gazebo Harmonic does not load with NVIDIA

javizqh, 2024-12-13:
Gazebo Harmonic does not load with NVIDIA, it stays in black

javizqh, 2024-12-13:
Can you take a look @dpascualhe ?

dpascualhe, 2024-12-14:
sure! I'll take a look. Just to confirm:
- Is the same setup working fine for Gazebo classic?
- Is acceleration with Intel / AMD / CPU working fine?

javizqh, 2024-12-14:
It works fine for Intel and CPU.

javizqh, 2024-12-14:
And launching with NVIDIA works for all the Gazebo Classic exercises

javizqh, 2024-12-14:
To launch it first use:
```bash
docker run --hostname my-postgres --name academy_db -d    -e POSTGRES_DB=academy_db     -e POSTGRES_USER=user-dev     -e POSTGRES_PASSWORD=robotics-academy-dev     -e POSTGRES_PORT=5432     -d -p 5432:5432     jderobot/robotics-database:4.7.0-beta
```

And when it finishes launch:

```bash
docker run --rm -it $(nvidia-smi >/dev/null 2>&1 && echo "--gpus all" || echo "") --device /dev/dri -p 6080:6080 -p 1108:1108 -p 7163:7163 -p 7164:7164 --link academy_db jderobot/robotics-academy:4.7.0-beta
```

javizqh, 2025-03-28:
The error is
` libEGL warning: egl: failed to create dri2 screen`

javizqh, 2025-03-31:
Change the 4.7.0 beta for latest and pull the latest docker image