JdeRobot/RoboticsAcademy #2958: Server does not fully terminate on quiting server

AshishRamesh, 2025-02-16:
I noticed  that on Ctrl + C to quit the server the terminal doesn't fully terminate / cleanup all process. Here's the log for reference 
```
 Xserver on :1 is running!
Xserver on :2 is running!
gzclient is running!
10:07:16 [MainThread  ] [INFO ] (my_app_logger)  Server Launched
10:07:16 [MainThread  ] [INFO ] (my_app_logger)  Visualization transition finished
10:07:16 [MainThread  ] [INFO ] (my_app_logger)  State changed to visualization_ready
^C
program exiting gracefully
10:07:30 [MainThread  ] [INFO ] (my_app_logger)  client disconnected: {'id': 1, 'handler': <manager.comms.websocker_server.WebSocketHandler object at 0x78038b98fb80>, 'address': ('172.17.0.1', 59996)}
10:07:30
INFO:manager.comms.websocker_server:Client asked to close connection.
10:07:30 [MainThread  ] [INFO ] (my_app_logger)  Terminating visualization launcher
10:07:30 [MainThread  ] [INFO ] (my_app_logger)  Terminating world launcher
```
Due to this i have to kill the terminal. Any subsequent tries to run the docker image results in the following error 

`docker: Error response from daemon: driver failed programming external connectivity on endpoint flamboyant_jang 
Bind for 0.0.0.0:7164 failed: port is already allocated.
`
To resolve this i delete the container and run a new container by running docker run . This means that every time i have to run the image i have to delete the image and run it again. Any idea whats causing this ?

i'm running the robotics-academy:latest image and only using the CPU.

javizqh, 2025-02-16:
Can you please share the command you are using to launch the docker image?

AshishRamesh, 2025-02-16:
`docker run --hostname my-postgres --name academy_db -d    -e POSTGRES_DB=academy_db     -e POSTGRES_USER=user-dev     -e POSTGRES_PASSWORD=robotics-academy-dev     -e POSTGRES_PORT=5432     -d -p 5432:5432     jderobot/robotics-database:latest`

`docker run --rm -it -p 6080:6080 -p 1108:1108 -p 7163:7163 -p 7164:7164 --link academy_db jderobot/robotics-academy:latest
`

javizqh, 2025-02-16:
The --rm flags should kill the image when it ends. It seems to be failing for you. I will look for an alternative way to kill the container

javizqh, 2025-02-16:
Try using:

javizqh, 2025-02-16:
Try using:
`docker ps -a`
To list all the docker images and then execute:
`docker rm CONTAINER_ID`
To remove the lingering container. This way you won't need to download the image every time

AshishRamesh, 2025-02-16:
yes this is the current method i'm using to get it working , just wanted to report the issues and find out if there was a better alternative as this has to be done every time i run the server.

javizqh, 2025-02-16:
We have been using the same command for almost a year an no errors have been reported related to this. Which docker version are you using and in what OS? 

With Docker version 27.2.0, build 3ab4256 in Ubuntu 24.04 this error does not occur

AshishRamesh, 2025-02-16:
Docker version 26.1.3, build 26.1.3-0 and ubuntu version is 22.04

javizqh, 2025-02-16:
Try to update docker to see if that solves it. I cannot help you any more than that.

Thanks for notifying this, if you ever solve it please share what you did here so that it can help other people

AshishRamesh, 2025-02-16:
Also on further testing i noticed that this happens only when i have the exercise open for example the follow_line or autoparking environment then when terminating the server results in the above mentioned issue but if i back out to the home page that is `http://0.0.0.0:7164/exercises/`and the quit then the server terminates as intended.

javizqh, 2025-02-16:
Thanks for that, I will look further into it

javizqh, 2025-03-17:
Can you @AshishRamesh please check if this keeps happening with the updated script in the current version?

AshishRamesh, 2025-03-18:
Updated to V5.7.0 and tested the same , the result is unfortunately the same.
`^C
program exiting gracefully
16:01:33 [MainThread  ] [INFO ] (my_app_logger)  client disconnected: {'id': 1, 'handler': <manager.comms.websocker_server.WebSocketHandler object at 0x7144245ac310>, 'address': ('172.17.0.1', 54706)}
16:01:33
INFO:manager.comms.websocker_server:Client asked to close connection.
16:01:33 [MainThread  ] [INFO ] (my_app_logger)  Terminating visualization launcher
16:01:33 [MainThread  ] [INFO ] (my_app_logger)  Terminating world launcher
`
Although now there is an additional issue when i try 
`docker ps -a` and then `docker rm CONTAINER_ID` it doent delete the container i instead get the following error :
`Error response from daemon: cannot remove container "/confident_jepsen": container is running: stop the container before removing or force remove`
i have to force remove it using `docker rm -f CONTAINER_ID` 
The workaround of closing the excersice and then terminating the server still works , it terminates gracefully.

meekhumor, 2025-03-18:
To avoid these issues and ensure proper process cleanup i would recommend you to use docker compose instead of running individual docker containers. 

Navigate to the main Robotics Academy directory and run `cp compose_cfg/user_humble_cpu.yaml docker-compose.yaml`

Start the application using docker compose:
`docker-compose up`

and then to stop the application:
`docker-compose down`

This ensures all running containers and processes are properly shut down, preventing port conflicts in subsequent runs.

jmplaza, 2025-03-18:
We have two ways of using RoboticsAcademy: (a) as a user, downloading the RADI from the Dockerhub, launching it locally and using it from the browser; and (b) as a developer, installing locally the repository, etc...

Both ways should work seamlessly.

javizqh, 2025-04-04:
Can you test again with the new RADI version? This should have been fixed

AshishRamesh, 2025-04-06:
Just tested v5.7.2 , the issue has been fixed. Thank You!

closed 2025-04-06 by AshishRamesh