JdeRobot/RoboticsAcademy #3009: Pick and Place exercise, just for Industrial Robots

jmplaza, 2025-03-20:
using Gazebo, ROS2 Humble and MoveIt2. 

@diegomrt has finished a [wonderful prototype](https://www.youtube.com/watch?v=XbqyYpIzRlE), but it is not yet integrated into RoboticsAcademy.

jmplaza, 2025-05-16:
Hi @ShauryaJain03 ,

@diegomrt and @javizqh are working on it and they are very close to an initial release of this exercise. The [prototype with MoveIt2 and ROS2 is working fine](https://www.youtube.com/watch?v=XbqyYpIzRlE).

We have some problem launching this Gazebo world inside RoboticsBackend (maybe @javizqh and @diegomrt may provide more details) and some betatesting of the initial release will be useful too. Stay tuned :-)

ShauryaJain03, 2025-05-16:
Sure, are there any other exercises that require beta-testing or some minor improvements?
I'd more than happy to contribute in any way that you may suggest, be it ROS2 or computer vision/machine learning related.
Do let me know!

javizqh, 2025-05-16:
The main problem we have is that the Gazebo classic instance that we are using inside the Robotics Backend (or RADI) is not able to launch the ros controllers properly. This causes the robotic arm to not work at all, so fixing this issue is really important right now. The only problem is that the documentation is non existant.

ShauryaJain03, 2025-05-16:
@javizqh I’ve previously worked with ROS2 controllers and Gazebo Classic and I’m comfortable debugging launch/setup issues. If it's okay with you, can I pick this issue up? 
Since the documentation is limited, I can also help create or update it as I work on it.
Thanks!

javizqh, 2025-05-16:
Sure, the documentation that is missing is from Gazebo Classic, so it is not ours, and as it is no longer maintained it cannot be updated in their repositories.

ShauryaJain03, 2025-05-16:
Thank you! I see, I thought there was some documentation that needed to be updated from your side. 
I went through the [contributing guidelines](https://github.com/JdeRobot/RoboticsAcademy/blob/humble-devel/CONTRIBUTING.md) and the [instructions](https://github.com/JdeRobot/RoboticsAcademy/blob/humble-devel/docs/InstructionsForDevelopers.md), can I proceed with cloning the repo and generating a RADI as given on the instructions page?

javizqh, 2025-05-16:
Yeah. Try to fork the repo instead of just cloning, it will probably be easier for you.

Also you will probably need to create a custom RADI if you want to change some dependencies. If you need some help, just write here and I will try to reply.

ShauryaJain03, 2025-05-16:
Alright thanks! I'll look into it and keep updating you.

closed 2025-06-13 by javizqh