JdeRobot/RoboticsAcademy #3046: Frontend build fails if src/exercises/ directory is missing

Im-Arth1307, 2025-04-21:
### **Description**
When running `npm run build`, the frontend build fails if the `src/exercises/` directory is missing. This is caused by a dynamic import in `src/exercise-index.js `(around line 70), which expects files to be present in that folder.
line:
``rendered_component = await import(`exercises/${component_path}`).then(...)``

While local development might work (e.g., when using dev servers or mounted volumes), production builds and CI/CD pipelines break, leading to confusion for new users and contributors.

I'm currently a GSoC 2025 applicant for this organization (awaiting results), and encountered this issue while trying to test UI changes and following the standard build process after cloning the repo.

### **Steps to Reproduce**

1. Clone the repository.

2. Navigate to react_frontend/ and run:
```
npm install
npm run build
```
3. Observe the build failure:
```
ERROR in ./src/exercise-index.js
Module not found: Error: Can't resolve 'exercises'
```

### **Why this matters**

- Build fails: Without the folder, `npm run build` doesn’t work.
- Inconsistent experience: Some local setups may work without a build, hiding the issue.

### **Suggested Fixes**

- Add a placeholder `src/exercises/` folder with a dummy file.
-  Improve `exercise-index.js` to handle missing files gracefully and warn instead of crashing.

Zouziszzm, 2025-04-22:
I have fixed this issues, and also found out another issue  " Replace broken static import of `common.zip`" in `PlayPauseButton.js`. so have fixed that as well.

Im-Arth1307, 2025-04-24:
@javizqh No, this issue doesn't arise while using `develop_academy.sh`, and the live-reloading experience worked well. The issue only came up when I tried a manual `npm run build` after cloning the repo to test a custom change. I thought it might still be worth flagging.

javizqh, 2025-04-24:
As said before we do not support local development methods outside of the ones stated in the documentation and will not change in the foreseeable future.

closed 2025-04-24 by javizqh