JdeRobot/RoboticsAcademy #3048: Build fails due to static import of common.zip (PR raised #3047)

Zouziszzm, 2025-04-22:
## Issue: Build fails due to static import of `common.zip`

### Description
The frontend build fails when attempting to statically import `common.zip` in `PlayPauseButton.js`:

```javascript
import commons from "../../common.zip";
```
Webpack cannot process .zip file imports directly, resulting in:

Module not found: Error: Can't resolve '../../common.zip'

### Impact
 -- Breaks production builds (npm run build)

 -- Incorrect asset handling (ZIP files shouldn't be JS imports)

 -- Confusing error for new contributors/CI systems

 -- Not documented in known issues

 ### Reproduction Steps

1. Clone the repository
2. Navigate to the project directory:
   ```bash
   cd react_frontend/
   ```
3. install
    ```
    npm install 
    npm run build
    ```
### Additional Context
Discovered while addressing #3046 (missing src/exercises/ bug)

Fix "included in fixes done to handel missing imports in both exercise-index.js and Pl… #3047" PR

Zouziszzm, 2025-04-22:
PR rasied for this fix "#3047"

javizqh, 2025-04-22:
Please make sure to read the [documentation](https://github.com/JdeRobot/RoboticsAcademy/blob/humble-devel/docs/InstructionsForDevelopers.md) on how to launch Robotics Academy first.

Can you please respond if after following those steps, either using docker run or our developer script, this issue is still happening to you.

closed 2025-04-23 by Zouziszzm