JdeRobot/RoboticsAcademy #3058: Basic Computer Vision extension for getting frames from videos at YouTube

jmplaza, 2025-04-30:


jessiffmm, 2025-05-04:
We're having trouble with the implementation. As a first step, we wanted to play the YouTube video in the same place where the camera feed is displayed. For the time being we had replaced the use of Camera with VideoYoutube. But we get the error:
Invalid hook call. Hooks can only be called inside of the body of a function component. This could happen for one of the following reasons:
1. You might have mismatching versions of React and the renderer (such as React DOM)
2. You might be breaking the Rules of Hooks
3. You might have more than one copy of React in the same app

We did an example in the folder prueba_youtube. This can be executed by putting: npx serve .
And then opening the page at the address http://localhost:3000

belal-bh, 2025-05-08:
Hi, is there anyone working on this issue or it has already been fixed ? I am interested to work on this issue.

codezerro, 2025-05-15:
Hi @belal-bh,

Thanks for your interest. We're now working on a [feature](https://github.com/JdeRobot/RoboticsAcademy/issues/3067) where users can upload/attach a local video on the browser to run a computer vision exercise.

Thakor-Yashpal, 2025-10-24:
Instead of directly replacing the camera component with a YouTube component, you should implement a single, flexible VideoPlayer component. This component will be responsible for deciding which video source (camera, local file, or YouTube) to render based on a simple prop, making the integration clean and less error-prone.

jmplaza, 2025-10-28:
Hi @Thakor-Yashpal , 

the use of react-player (https://www.npmjs.com/package/react-player) sounds really good to me. As long as we can get periodic frames in the browser from the YouTube (and vimeo...) video player we could send them to the robotics application running in the RoboticsBackend in the same way we do for real cameras (the frames are obtained from the webRTC sw in that case).

Are you up for doing a prototype? Just a single static webpage prototype (a) getting one frame from react-player sw (and a demo YouTube URL) and (b) showing that at a that static webpage. In order to reduce complexity for now, not integrated in RoboticsAcademy frontend yet.  That initial prototype could be improved later for getting+sending not only a single frame, but many of them in loop, continuous, periodic way.

Thakor-Yashpal, 2025-10-29:
Hi @implaza,

Thank you for laying out the plan. The idea of starting with a static webpage prototype using `react-player` to get the frames sounds like a great approach.

I'm happy to work on this. I'll be able to start on Friday and will post my progress here.

buildwithsoumya, 2026-10-11:
Hi @jmplaza, I opened #4044 implementing direct video URL support in the Video tool, following your suggestion (frames captured in the browser and sent through the existing `pick` pipeline). YouTube page links aren't supported since their iframe can't be read by a canvas; details and test results are in the PR. Feedback welcome!