JdeRobot/RoboticsAcademy #3083: Migrate frontend to Typescript

javizqh, 2025-05-23:
Migrate frontend to Typescript and remove all outdated parts

codezerro, 2025-05-23:
Great initiative.

javizqh, 2025-08-06:
We are already working on It. Sorry

jmplaza, 2025-08-06:
Nice to meet you @aditi25srivastava , 

no explicit assignment is required. If you want to contribute, I suggest you first get familiar with RoboticsAcademy as user, after that get familiar with its software base (https://github.com/JdeRobot/RoboticsAcademy/tree/humble-devel/docs, https://github.com/JdeRobot/RoboticsAcademy) as developer. 

Maybe the first step for you is solving some exercises of our RoboticsAcademy, for instance the FollowLine (you will learn visual control) and BasicVacuumCleaner (you will practice with reactive FiniteStateMachines). Just solve them using the latest release of RoboticsAcademy and the hal.py Python module to access to robot sensors and actuators. Share here the corresponding videos for your solutions working. After that you may solve them again but without using hal.py module, just directly using ROS2 topics.

aditi25srivastava, 2025-08-06:
Thank you for your guidance. I’ll start by getting familiar with RoboticsAcademy and will begin with the FollowLine or BasicVacuumCleaner exercise. I’ll share a video once I’ve completed it. Looking forward to contributing more!Thank you for your guidance. I’ll start by getting familiar with RoboticsAcademy and will begin with the FollowLine or BasicVacuumCleaner exercise. I’ll share a video once I’ve completed it. Looking forward to contributing more!

closed 2025-09-08 by javizqh